universam1/iSpindel #363: ispindel

james15071982, 2020-03-29:
i have flashed with the bin file and now im not getting any signal on wifi so I tried reflashing it back to factroy restore and seeing if I got a signal but nothing PLEASE HELP

ErikdBr, 2020-03-29:
Did you build the whole thing? Otherwise it will not work.

thegreatgunbantoad, 2020-03-29:
Might be teaching granny to suck eggs here, but, you do know that once if has your WiFi credentials it will go to sleep and make up every 15 mins (configurable), to bring it out of that and serve it's portal you need to double tap the reset contacts/button.

james15071982, 2020-03-29:
I did build the whole thing and have tried tapping the reset button and it still doesn’t work , the blue light just Flashes five times then goes off

ErikdBr, 2020-03-30:
Try what is in the FAQ: https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md
where it says: "What should I do if my iSpindel isn't working properly after flashing it with the firmware?"

kordi83, 2020-04-04:
James did you figure it out i have the same problem

closed 2020-06-23 by stale[bot]