universam1/iSpindel #364: After firmware flash, MCU is not showing the iSpindel AP when pressing the reset 4-5 times

peterd0404, 2020-04-03:
I have 3 boards.  1 one them works but the other 2 do not.  Once firmware flash is complete, I press the reset button to go into Configuration mode.  But I do not see the iSpindel AP.  What I am seeing is the an "ESP_xxxxxx" AP. 
Initially, I flashed the firmware using Windows ESP8266 flasher tool.  Then I started exploring flasher for my Mac. I tried flashing using MacOS ESP Flash tool, ESPtool, and ESP8266 Flasher for Windows.  Still the same result.
I have erased the flash and reflashed v6.2.  I have also tried v6.1.1 still the AP does not change.  

Below is the export of the flash on my Mac Terminal:

**python ./esptool.py --port=/dev/cu.usbserial-1420 write_flash 0x00000 firmware.bin
esptool.py v3.0-dev
Serial port /dev/cu.usbserial-1420
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:f4:32:2d:af:63
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 454288 bytes to 313485...
Wrote 454288 bytes (313485 compressed) at 0x00000000 in 30.4 seconds (effective 119.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...**

At this point, I think two of my MCUs are a different version.  It does not make sense that one works and other two does not.  

Any thoughts or insight is greatly appreciated.  

Thank you for your time.
-Peter

peterd0404, 2020-04-03:
Just an update I saw an issue #324 and after flashing to firmware v5.8.6.  I now see the iSpindel AP.  Now I am going through the FAQ to initialize memory to load 6.2.  I will report back later

peterd0404, 2020-04-03:
I have tried the FAQ to initialize the memory but v6.2 and v6.0.6 still does not work.  It looks like I will be at v5.8.6 on these two boards.  Unless others have other suggestions.

Jasbenco, 2020-05-07:
I've got the same issue. Tried it on 3 different levels of Wemos d1 minis, a 3.1.0, a D1 mini Pro v1 and a v2 D1 mini. I have tried the FAQ to initialize the memory with no change. Glad I spotted your post I had been going round and round in circles for a couple of days trying to flash 6.2 even to the point of repurposing my PI 4 home assistant into a standard Linux box

jrventer, 2020-05-14:
I had the same issue and found when I connected serial that the software was not able to detect the sensors. Seems like with the 6.x.x versions if the sensors are not detected the config mode does not work properly. With 5.8.6 it worked while the sensor was not detected. Once I fixed the loose connection config mode in 6.2.0 worked without an issue.

Jasbenco, 2020-05-14:
Thanks, I'll set one up on a breadboard and test it out. I'll post up the results later.

Fatfil, 2020-05-20:
All I can confirm this is the case. I have just built my iSpindel and while testing it prior to stuffing it in the tube I could not get the iSpindel AP to appear. Flashed with 5.8.6 and it worked but the temp sensor wasn't working. Checked the board and I'd wired the DQ of the 18B20 to the D5 pin of the D1 rather than D6. Fixed this, reflashed with 6.2.0 and hey presto everything is working. Thanks for the pointer jrventer I was starting to think I'd fried the D1 during the build.

closed 2020-07-26 by stale[bot]