universam1/iSpindel #365: Problem with build

kordi83, 2020-04-04:
Hi,

I have build 2 different ispindels with 4.0 board  Joey Joe board. When i flash it with 6.2 i can`t get it in config mode. When i use 5.8 it can`t safe any info. in the info shows that my temp sensor is not found and it is at -127. I have reordered and used sensor form another supplier.  Please help

mnandres, 2020-04-28:
Hi, i have a similiar problem. some help too find the problem?

thegreatgunbantoad, 2020-04-28:
It's highly unlike that the temperature sensor is not working.  I've never had one fail, bad gyros and poorly soldered Wemos boards I have.  Check the joints on the Wemos' daughter board.  Also check that the temp sensor is fitted correctly to the correct side of the board.  It does look though like you have some other underlying issue.

LloydPickering, 2020-05-06:
I am having exactly the same problem, except I have plugged everything except battery charger into a breadboard to diagnose (except the GY521 which is already soldered on). I have continuity checked the connections where possible and it all seems fine. I have tried reversing the temp sensor.

Picture of breadboard attached. I've tried to use the same colour scheme as the circuit diagram for clarity. [image]

When flashing 6.2 I get the same messages as the original poster.

When flashing 5.8 I can go in and edit the config. I can see the angle of inclination seems to be working, temp sensor is not, showing -127 (I guess it defaults to min value on a signed char). When I save settings, the output complains the config is corrupt and resets to being an AP. Here is the log from that (replacing the wifi ssids with [redacted]):

FW 5.8.6
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
...........
           ERROR no Wifi crenals

goingto Config Mode
ERROR: OW DISCONNECTED
ERROR: OW DISCONNECTED
*WM: Scan done
*WM: DUP AP: [redacted]
*WM: DUP AP: [redacted]
*WM: DUP AP: [redacted]
*WM: DUP AP: [redacted]
*WM: DUP AP: [redacted]
*WM: DUP AP: [redacted]
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding aameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring accs ont..
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
ERROR: OW DISCONNECTED
*WM: Request redirected to pve▒▒х▒5
                                   ERROR: OW DISCONNECTED
ERROR: OW DISCONNECTED
*WM: Request redirected t▒▒ѥٕ portal
ERROR: OW DISCONNECTED

closed 2020-07-24 by stale[bot]