universam1/iSpindel #367: ISpindel stops working after about 3 Minutes

CBBC2020, 2020-04-16:
I’ve built an iSpindle from a kit. Flashed using Nodemcu and all appeared fine. iSpindle appears on the network list and I can connect and configure it. ISpindel information is displayed ie. temp, tilt etc and the  Blue LED is flashing every second. After about 3 minutes the LED flashes brightly once and the iSpindle disappears from the network list. Pressing reset kicks it off again but then stops in the same way. Help anyone.

mcmellow, 2020-04-16:
What value is your timer interval set?

CBBC2020, 2020-04-16:
Hi, I take it you mean the interval for sending data to Brewfather, this is set at 900 secs.

mcmellow, 2020-04-16:
So it is not getting awake after the first interval (900 sec).  It is earlier.  Did you check all soldering and are all the components are mounted correctly.  You could check this side https://www.esp8266.com/

CBBC2020, 2020-04-16:
It is only running for 3 minutes and then stops. I would assume that if any of my circuitry is wrong then I would not be able to flash and it wouldn’t run at all.

CBBC2020, 2020-04-16:
Ah, didn’t see the last bit of your reply. I will check.

mcmellow, 2020-04-16:
See closed issue #343  
My iSpindle worked correctly once and fell in a deep sleep. Only switch on and on trigged once. Cause a incorrect mounted diode. Mounted it correctly and soldered again. It worked like a charm. When the wemos board is OK, but the connections or components on the mainboard are not correct. You are able to flash, but working can give unpredictable results.
Success with the checking

CBBC2020, 2020-04-17:
I’ve disconnected the circuitry from the Wemos and it is doing exactly the same thing. I’ve down loaded Putty and run it on Com 3 but I don’t really know what the output is telling me. The first image is when it’s running and outputs every time the LED blinks. The second is when it stops!

[image]

[image]

mcmellow, 2020-04-17:
You have disconnected the Wemos. When you did flash your Wemos, did you do this with clear all memory first?
Sorry, can not make any sense of your screenprints.

CBBC2020, 2020-04-17:
I know, just looks like rubbish. Yes I flashed Wemos to clear all memory first as I thought it looks like it is running out of memory then stopping. Thanks for your interest and feed back. I wait in hope that someone out there has had the same problem and knows how to fix it.

mcmellow, 2020-04-18:
Just check my 2 iSpindels.  This looks your configuration file is not loaded (or present) because nothing happens. No worker started.Atleast you should see somthing started with this:

[image]

Do a configuration, ie just the time interval or so and make sure you hit the save button and wait until it is fully saved.
Or start with a new flash of the latest firmware 6.2.0 or later with a wipe of all data.  And do an configuration after the new flash.

Success

CBBC2020, 2020-04-18:
Being very much a novice at this, I have come to the conclusion that the iSpindel is actually working correctly. It is simply going into sleep mode after a few minutes and then attempting to sent data at every set interval. The problem I now have is that I am trying to sent data to Brewfather and it is not connecting. Have uploaded a screen shot so any help would be appreciated.
[image]

mcmellow, 2020-04-18:
As far as I can see it is working correctly. About the connection to Brewfather, I know nothing. Have look at the brewfather site for this.

universam1, 2020-04-18:
Clearly the log says „failed to connect “ so your server is not reachable

closed 2020-04-29 by CBBC2020