universam1/iSpindel #371: Cannot see iSpindel AP after recharging battery.

CBBC2020, 2020-05-02:
ISpindel has been working 100% for 3 weeks. Charged the battery today and now cannot see the iSpindel AP. It is still updating Brewfather every 15 minutes but pressing the reset button a few times does not give me the blinking blue led so cannot access it to configure. So the only thing I have done is switch off the Wemos, connect mini usb to TP charger, red light on charger blue light when charged. Switched Wemos on and press reset a few times. I now get the following sequence on the Wemos - 1 blink, 2 blinks, 2 blinks 1 longer blink and nothing. Only difference is that the fully charged battery measures 4.2 v. Any ideas. TIA.

RonnyRusten, 2020-05-04:
Keep trying the reset button. 3 or 4 rapid klicks on the reset should work. It can sometimes be a little tricky.

samdeetwo, 2020-06-14:
I have a similar thing: calibrated the tilt then tried to access AP again, now Blue light flashes twice fast, then 3-4 times regular, at which time iSpindel AP appears, but then jumps a few seconds and flashes twice fast. This WiFi AP jumps and disappears and I cannot access the settings page.