universam1/iSpindel #376: Ubidots alternative

guilleiwd, 2020-05-13:
Hi every one. A pleasure to join.
I have a doubt. There is any alternative to ubidots to check the fermentation and other data. Because ubidots told me something about my maximum dots per day and I don't get new information. Kind regards.

thegreatgunbantoad, 2020-05-13:
Brewfather rules.  It's not free but it is fairy cheep.

thegreatgunbantoad, 2020-05-13:
Brew spy is free, but it's a bit meh

ErikdBr, 2020-05-13:
I sur brewspy and it's great! But it is an app on your phone and is not free.
[image]

guilleiwd, 2020-05-13:
Thanks very much for the comments mates. I will check both. A big hug from Spain!!

petjez, 2020-05-16:
Hey. I use monitor.beer. It's a free website and just simple to use.

Patrick

guilleiwd, 2020-05-16:
I will check it as well!  Thanks mate!!

closed 2020-07-22 by stale[bot]