universam1/iSpindel #379: Can't save new formula!

Naesstrom, 2020-05-19:
When I try to save the new formula (-0.001344537*tilt^2 + 0.616671894*tilt - 13.73642293) it reboots and then the original formula is still there. Everything else seems to be working and the log says the config is saved?
```
saving config...
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.002083958*tilt^2 + 0.737917168*tilt - 18.06203205","SSID":"","PSK":"","Offset":[0,0,0,0,0,0]}
saved successfully
```
The problem is that it then reboots 2 times and seems to save the standard config again?
```
saving config...
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"","SSID":"","PSK":"","Offset":[0,0,0,0,0,0]}
saved successfully
Samples:30 min:78.07 max:78.17 time:520
x: -3166 y: 14064 z: -590
Tilt: 78.12
Tacc: 24.22
Volt: 4.43
Temp: 22.87
Parse error at 1
Gravity: 0.00
After waiting 0ms, result 0
Rescue Wifi credentials
failed to connect
```
Only thing I can find in the log that could indicate some kind of problem is the `deserializeJson() failed: NoMemory` in the boot. Tried reinstalling and clearing memory etc.
```
FW 6.2.0
2.2.1(cfd48f3)
Worker run!
mounting FS... mounted!
reading config file
deserializeJson() failed: NoMemory
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 C8 D6 16 A8 1 3C D9
  Chip = DS18B20
  Data = 1 70 1 55 5 7F A5 A5 66 C7  CRC=C7
  Temperature = 23.00 Celsius, 
offsets not available
Boot-Mode: Deep-Sleep Wake

Double Reset detected
...........
ERROR no Wifi credentials

going to Config Mode
```

thegreatgunbantoad, 2020-05-22:
Don't forget the multiplication symbol * also take out spaces and add some more brackets to be safe, should take the form: (1.0000E-6*(tilt^3))+(-1.3137E-4*(tilt^2))+(6.9680E-3*tilt)+(9.0206E-1)
You don't have to use E notation but it helps keep things under the 100 character limit.  Also a 3rd order polynomial is usually a better fit.

Naesstrom, 2020-05-24:
Don't think the formula is the problem, used the excel sheet and used the same on other iSpindels without editing them

gingerneil, 2020-06-23:
Have you tried 6.3.1 ? I had an issue of config not saving and the same error, and it resolved with the firmware changes- I think the increase of the JSON size did it.

closed 2020-08-29 by stale[bot]