universam1/iSpindel #383: DynamicJsonDocument doc(size * 2)

gingerneil, 2020-06-02:
Pulling this upwards from a previous comment/issue...
I seem to be hitting this problem on all the gyros I've recently bought... I appreciate thats a hardware issue, and changing the software isn't ideal. However, as several people seem to be having this issue, would it be possible to patch the next release with (size * 3) ? Does this negatively impact anything else ?
For now, I have reverted back to the last main 5.x release, pre-JSON, and that seems to be working ok but it means I miss out on 6.x improvements.

universam1, 2020-06-09:
@gingerneil would you like to make a PR proposal?

gingerneil, 2020-06-13:
I'm fairly new to git, and not a developer, but will try! At least it seems a simple change!

thegreatgunbantoad, 2020-06-14:
@gingerneil Did you manage to do a pull request?  When I tried to do one, the button was inactive.

gingerneil, 2020-06-14:
Hi,
Yes - just done it now... First timer!

gingerneil, 2020-06-15:
Added into 6.3.0

closed 2020-06-15 by gingerneil