universam1/iSpindel #384: Temperature units don't select as expected

thegreatgunbantoad, 2020-06-07:
When selecting the temp units in the in the configuration, they only seem to change if you select twice before saving.  Moving the sTS; call out of the window.onload in to a window.onchane function seems to resolve it.

In WiFiManagerKT.h it would then look like:
...
window.onload = function (e) {
  for (var i = 0; i < labels.length; i++) {
    if (labels[i].htmlFor != '') {
      var elem = $(labels[i].htmlFor);
      if (elem) elem.label = labels[i];
    }
  }
  value = $('selAPI').value;
  sAPI(value);
  $('TS').value = $('tempscale').value;
  fillopt();
  $('API').querySelector('option[value="'+value+'"]').selected = true;
}
window.onchange = function (e) {
  sTS();
};</script>)V0G0N";
...

universam1, 2020-06-09:
@thegreatgunbantoad Thank you for the feedback - would like to create a PR for?

thegreatgunbantoad, 2020-06-10:
I've been playing with adding temperature compensation to the gravity calc, and needed to have a drop down option to enable or disable it.  After a day of struggling with why I couldn't get the value to stick I eventually noticed that it was all sat in an onclick event.  To that end is there currently a way to calibrate the temperature sensor?

thegreatgunbantoad, 2020-06-14:
Yay update approved!

closed 2020-06-14 by thegreatgunbantoad