universam1/iSpindel #389: Access point instable

samdeetwo, 2020-06-15:
Hello

We built two iSpindels on Sunday, and both have a problem of instability of the access point. 
For the first one we managed to get a stable access point, calibrated the tilt, then tried getting back to the access point and it started being unstable. Basically the blue light blinks twice rapidly then three times at the regular intervals (for AP), at which time it is possible to connect to the iSpindle AP, but it then proceeds to skip a blink (we lose AP connection), and the sequence starts once more.
Any hints as to how to proceed?
Thanks!

universam1, 2020-06-15:
As usual without a console output there is no chance to give any solid help

samdeetwo, 2020-06-15:
Here is a log
{dl▒▒|▒l▒|
▒
l▒
c|▒▒▒s▒c▒c▒▒oo▒lgg▒▒▒
cp▒▒drlslp▒o▒
▒
l
▒▒
bo▒|$▒
▒c▒▒gg▒l▒▒d`▒o
l`gs▒▒▒nc▒l{▒▒g
b
▒o▒d▒x▒▒▒
   FW 6.0.5
3.0.0-dev(c0f7b44)
Worker run!
mounting FS... mounted!
reading config file
applying offsets
parsed config:
{"Name":"YannSpindel01","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"UR

L":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"0

00","Vfact":191.8,"TS":0,"OWpin":-1,"SSID":"Freebox_YLR","PSK":"Yann_et_Angelica

","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-3663,"aY":1713,"aZ":1653}
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
No devices found!
scanning for OW device on pin: 116
No devices found!
scanning for OW device on pin: 121
No devices found!
ERROR: cannot find a OneWire Temperature Sensor!
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:30 min:64.26 max:64.29 time:539
x: 9990 y: 12548 z: 420
Tilt: 64.28
Tacc: 29.93
Volt: 3.58
Temp: 0.00
Gravity: 20.47
After waiting 5982ms, result 6
Rescue Wifi credentials
failed to connect

Final-sleep: 900s; RT: 7886

Seems like the temp sensor doesn’t work. We tried another still no luck.

universam1, 2020-06-15:
It will not work without the necessary sensors, that needs to be fixed essentially