universam1/iSpindel #391: reset page not found

jonnygoldengonads, 2020-06-16:
In the Wifimanager 'Information' page, there is a section called 'Available Pages', which has a link '/r' which is meant to delete wifi configuration, but actually gives a 404 not found when clicked.

This appears to be in \pio\lib\WiFiManagerKT\WiFiManagerKT.cpp, where line 945 defines this link: 

page += F("<tr><td><a href=\"/r\">/r</a></td>");

Garywoo, 2021-01-17:
It's a shame this issue was closed by the bot, since it's a genuine bug. Can it please be reopened?

thegreatgunbantoad, 2021-01-17:
@Garywoo if you do want to delete your wifi config, the reset function will do it.

It actually seems to arise from the fact that /r has no binding
in the section following /* Setup web pages: root, wifi config pages, SO captive portal detectors and not found. */
and no handling routine that would do the function.  I wonder if it was removed at some point and one bit of code got missed out.

Garywoo, 2021-01-18:
@thegreatgunbantoad Unfortunatly reset function also doesn't delete the wifi config. See #372

ErikdBr, 2021-01-18:
It should after a couple of tries.

thegreatgunbantoad, 2021-01-18:
@Garywoo it reformats the memory and so wipes out the whole config file, so it should do.  But I have had units that that didn't work on, presumably due to poor quality ESP boards.

Not tried them but these look legit: https://hobby-store.co.uk/arduino/wemos-d1-mini-pro-16m-v10.html