universam1/iSpindel #395: Exception 23 in boot sequence

Takatu, 2020-06-24:
## I'm submitting a ...
  - [x ] bug report
  - [ ] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report
I have two M1 mini. One I can flash and boot up to configuration mode without any issues but the second one is throwing Exception 23 in the boot sequence.

### Console Logs
sd<\0>dœŸ|<\0>„là|<3><4><\f><\f>„<4>lä<4>c|‡ƒ<3>ì<19>›{’c„<\f>c„ûggŸlg'œãä<\f>b<28>pŒŽdsd{lpûgà<16><3><\f><\f>ƒ<\f>l<4>Äœ<\f><4><4>c<4>nã|<2>ä<4>‡sÜ<4>cŒónoï<\0>dÄ‡l`<3><27><19>'o<\f>l`<3><15><2>'{Û“n<4><\f><3><4>‡<15>d<14>{Û“n<4><\f><3><4>‡<15>dœ<27>Û<\f><28>ŒŒŽ<28>“l`<3>ü‚gœ<3><\n>FW 6.3.1<\r><\n>
2.2.2-dev(38a443e)<\r><\n>
Worker run!<\r><\n>
mounting FS... mounted!<\r><\n>
ERROR: failed to load json config<\r><\n>
<\n>ERROR config corrupted<\r><\n>
scanning for OW device on pin: 5<\r><\n>
No devices found!<\r><\n>
scanning for OW device on pin: 12<\r><\n>
No devices found!<\r><\n>
scanning for OW device on pin: 116<\r><\n>
No devices found!<\r><\n>
scanning for OW device on pin: 121<\r><\n>
No devices found!<\r><\n>
ERROR: cannot find a OneWire Temperature Sensor!<\r><\n>
Acc Test Connection ERROR!<\r><\n>
offsets not available<\r><\n>
Boot-Mode: Deep-Sleep Wake<\r><\n>
...........<\n>ERROR no Wifi credentials<\r><\n>
<\n>going to Config Mode<\r><\n>
Acc Test Connection ERROR!<\r><\n>
<\r><\n>
Exception (28):<\r><\n>
epc1=0x40209c1f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000<\r><\n>
<\r><\n>
>>>stack>>><\r><\n>
<\r><\n>
ctx: sys<\r><\n>
sp: 3ffffd60 end: 3fffffb0 offset: 0190<\r><\n>
3ffffef0:  00000000 00000000 00000000 402026a0  <\r><\n>
3fffff00:  0031803d 3ffef3a0 3ffefdb8 4020a39a  <\r><\n>
3fffff10:  00000000 00000000 3ffef9a8 4020253c  <\r><\n>
3fffff20:  0031803d 3ffef3a0 3ffefd9c 40202e84  <\r><\n>
3fffff30:  003174dd 3ffef3a0 3ffef378 40223766  <\r><\n>
3fffff40:  40253b18 3ffefd9c 3ffef3a0 40223e0a  <\r><\n>
3fffff50:  60000600 3ffefd9c 3ffef3a0 40253b25  <\r><\n>
3fffff60:  40253b6a 3fffdab0 00000000 3fffdcb0  <\r><\n>
3fffff70:  3ffef3b0 3fffdab0 00000000 00000004  <\r><\n>
3fffff80:  40000f49 40000f49 3fffdab0 40000f49  <\r><\n>
3fffff90:  40000e19 00000005 00071664 00000000  <\r><\n>
3fffffa0:  00000000 aa55aa55 000000f6 40104c85  <\r><\n>
<<<stack<<<<\r><\n>
<\r><\n>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)<\r><\n>
<\r><\n>
load 0x4010f000, len 3456, room 16 <\r><\n>
tail 0<\r><\n>
chksum 0x84<\r><\n>
csum 0x84<\r><\n>
va5432625<\r><\n>
~ld<\n><\n>FW 6.3.1<\r><\n>
2.2.2-dev(38a443e)<\r><\n>
Worker run!<\r><\n>
mounting FS... mounted!<\r><\n>
ERROR: failed to load json config<\r><\n>
<\n>ERROR config corrupted<\r><\n>
scanning for OW device on pin: 5<\r><\n>
No devices found!<\r><\n>
scanning for OW device on pin: 12<\r><\n>
No devices found!<\r><\n>
scanning for OW device on pin: 116<\r><\n>
No devices found!<\r><\n>
scanning for OW device on pin: 121<\r><\n>
No devices found!<\r><\n>
ERROR: cannot find a OneWire Temperature Sensor!<\r><\n>
Acc Test Connection ERROR!<\r><\n>
offsets not available<\r><\n>
Boot-Mode: Exception<\r><\n>
...........<\n>ERROR no Wifi credentials<\r><\n>
<\n>going to Config Mode<\r><\n>
reboot RFCAL<\r><\n>
rl<\0>lœŸ<<\0>Œlà|<2><4><4><\f>Œ<4>$ì<\f>b|‡‚<3>ä<27>“s›blcŒóooßdogœãì<\f>c<28>pÄ‡l;lsd<27>ƒ;ò›oàcãnc8<\0>ü

## Steps to Reproduce (for bugs)
1. Connect to Windows
2. Followed the FAQ steps: What should I do if my iSpindel isn't working properly after flashing it with the firmware
3. Flashed firmware 6.3.1
4. Observe the HTerm when pressing the reset button

universam1, 2020-06-24:
There is no support running the firmware without the sensors, this will never work

closed 2020-06-24 by universam1

vytux-com, 2020-06-24:
Check the connections to the temp module, the error is telling you that the system cannot talk to it