universam1/iSpindel #397: iSpindel (Jeffrey) - only one reading on reset, never wakes from Deep Sleep

neilbaldwin, 2020-06-27:
I've just built an iSpindel (Open Source Brewing 'Jeffrey' to be precise) and am having an issue with it waking from Deep Sleep. It only outputs readings if I manually reset it. Once it goes back to sleep it never wakes again, unless I manually reset it. The Jeffrey PCB uses a BAT43 diode between D0 and RST.

I've monitored the voltage at both RST and D0 and they both stay high (3.3v) all the time. I might be misunderstanding but shouldn't D0 go low during sleep? Or is it the other way?

I've also tried shorting D0 and RST but this still does not work. The Jefffrey PCB has two pins that when bridged will cause a reset of the D1 Mini - this works.

I'm wondering if the D1 Mini is faulty. Is there anything I can test further? Terminal output (from Cool Term) below when I manually reset.

Cheers,
Neil

Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:64.15 max:66.19 time:757
x: 9812 y: 13206 z: -1734
Tilt: 64.93
Tacc: 28.08
Volt: 4.07
Temp: 26.44
Gravity: 20.80
IP: 192.168.8.16

calling HTTP
{"name":"iSpindel000","ID":16578179,"angle":64.92694,"temperature":26.4375,"temp_units":"C","battery":4.07195,"gravity":20.8035,"interval":5,"RSSI":-81}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 5s; RT: 1953
;l.dúﬂ|.ål‡<....Ñ.l‰.b<áÉ.‰.ì{ìcå.cÑ˚'oﬂ$gnú„Ï.#.xåèl;l{l

ErikdBr, 2020-06-27:
You know the bat43 has a + and - side as opposed to a resistor so maybe it is the wrong way around?

neilbaldwin, 2020-06-27:
> You know the bat43 has a + and - side as opposed to a resistor so maybe it is the wrong way around?

Yeah sure, it's definitely orientated correctly (as per the silk screen at least).

thegreatgunbantoad, 2020-06-27:
Reset lines are generally active low so as to be intrinsically safe.  So a 3.3v reading means you are NOT in reset.
Your iSpindel, with a standard setup will only wake up every 15mins.
Check your battery voltage reading, it should be 4.1v ish.  A good charge of the battery might resolve your issue, low battery causes weird stuff to happen.

neilbaldwin, 2020-06-27:
> Reset lines are generally active low so as to be intrinsically safe. So a 3.3v reading means you are NOT in reset.
> Your iSpindel, with a standard setup will only wake up every 15mins.
> Check your [...]

Ah OK, so reading 3.3V would indicate it being in sleep then, thanks.

Battery voltage is slightly over 4V (according to the info page on the iSpindel AP menu)

thegreatgunbantoad, 2020-06-27:
Simplest way to be sure is to monitor that line when you are talking to the portal.
The D0 to RST connection I believe is to allow the device to cause it's own reset.  I can't see on the data sheet any pin that explicitly indicates a deep sleep.  Reset would not likely be used with respect to a sleep function.
The blue LED will go off in deep sleep as it is actually connected to one of the comms lines, no comms, no LED.  You'd usually see it blip at it goes round checking it's sensors.

neilbaldwin, 2020-06-27:
Yes. From my understanding, the D1 shuts down leaving the RTC to trigger when the current timer value has elapsed causing D0 to send a signal to the RST pin so that the D1 wakes, spits out readings then shuts down again, and so on.

Monitoring the terminal output, once the D1 goes to sleep, it stays that way as there is no further terminal output. Could the BAT43 have failed? What would be the effect of that? Do they fail open or shorted?

thegreatgunbantoad, 2020-06-27:
Last time I fried a diode it went short.  Easy way to test that is to get into AP and instigate a reset.  If you lose the iSpindel WiFi, the reset is probably working.  I think saving config causes a reset.

neilbaldwin, 2020-06-27:
> Last time I fried a diode it went short. Easy way to test that is to get into AP and instigate a reset. If you lose the iSpindel WiFi, the reset is probably working. I think saving config causes a reset.

Ah, good idea. I will try that but I'm already very certain that it works (from the AP). Which is even more odd. Would that mean the RTC isn't working in terms of triggering the wake signal?

sergioguimaraes, 2020-06-27:
Is it a DOT.AM ESP8266? I have noticed that this manufacturer has some issues.

neilbaldwin, 2020-06-27:
> Is it a DOT.AM ESP8266? I have noticed that this manufacturer has some issues.

No. It just says "ESP8266MOD".

neilbaldwin, 2020-06-27:
> Last time I fried a diode it went short. Easy way to test that is to get into AP and instigate a reset. If you lose the iSpindel WiFi, the reset is probably working. I think saving config causes a reset.

This is what the terminal output looks like in AP mode. Weirdly I don't think it does reset on saving. You can see the saving output and you can see it changes the WiFi but it seems it doesn't actually reset. Unless I'm missing something:

`Double Reset detected

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: HUAWEI-B315-77A7
*WM: -67
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: iSpindel001
*WM: Parameter
*WM: sleep
*WM: 5
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: tempscale
*WM: 0
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 3
*WM: Parameter
*WM: token
*WM: 
*WM: Parameter
*WM: server
*WM: log.brewfather.net
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: channel
*WM: 0
*WM: Parameter
*WM: uri
*WM: /ispindel?id=wT6DHmM5YN5acU
*WM: Parameter
*WM: db
*WM: ispindel
*WM: Parameter
*WM: username
*WM: 
*WM: Parameter
*WM: password
*WM: 
*WM: Parameter
*WM: job
*WM: ispindel
*WM: Parameter
*WM: instance
*WM: 000
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.28
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...
{"Name":"iSpindel001","Token":"","Sleep":5,"Server":"log.brewfather.net","API":3,"Port":80,"Channel":0,"URI":"/ispindel?id=wT6DHmM5YN5acU","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"HUAWEI-B315-77A7","PSK":"sumisumi","Offset":[0,0,0,0,0,0]}
saved successfully
Samples:38 min:66.42 max:66.52 time:665
x: -7778 y: 14030 z: 2302
Tilt: 66.49
Tacc: 25.37
Volt: 4.05
Temp: 23.75
Gravity: 21.61
IP: 192.168.8.16

calling HTTP
{"name":"iSpindel001","ID":16578179,"angle":66.48821,"temperature":23.75,"temp_units":"C","battery":4.051095,"gravity":21.60952,"interval":1,"RSSI":-85}HTTPAPI: posting
code: 200
{ "result": "ignored" }

Final-sleep: 1s; RT: 52302
sd.$‹ü<.ƒd‡|....ƒ.$Ï.c|è√.Ï.€sõbÑ.#åÛogülngú„‰.c.pƒ«ds$sd`

neilbaldwin, 2020-06-28:
Hmmm. Definitely some fault on the hardware I think. I've just built another one (I have 5) and it's working fine.

neilbaldwin, 2020-06-28:
Ended up breadboarding the circuit so I could test the remaining D1 Mini boards. Out of the five I bought, 3 of them are faulty! One wouldn't power on at all, the one causing the issues in this thread definitely does not set a signal from D0 to perform the self-reset wake. So frustrating, there's so many poor quality clones on Ebay.

thegreatgunbantoad, 2020-06-28:
Try reflowing the joints on the metal hybrid device to the Wemos.  Like you say, there is some rubbish out there at the moment.  It can be hit and miss.

closed 2020-09-03 by stale[bot]

ErikdBr, 2020-10-14:
I build one today and have  the same issue, so a warning, don't buy these: https://nl.aliexpress.com/item/32832024575.html