universam1/iSpindel #401: Connection to Fermentrack

worgz01, 2020-07-08:
## Report

I am having issues continuing to fermentrack. I thought it was the fermentrack but I have exhausted that line of attack

[image]

I have had one success in connection after I improved the network strength but them no movement in data readings after that.

It does however connect successfully to UBIDOTS with no problem

I have configured correctly, I think, but cannot get any connection data to fermentrack
[image]

brendongough, 2020-07-12:
Path/URL looks wrong, try making it http://192.168.0.17/ispindel

worgz01, 2020-07-13:
Sorry, I tried what you suggested, as indicated in the first file snip. The second file was a previous snip before i changed the path setting. My bad.

[image]

closed 2020-11-02 by stale[bot]