universam1/iSpindel #402: Calibration

worgz01, 2020-07-13:
Second build and another issue, I have offset calibrated successfully (89.5 degree) but went to check 25 degree and indication were way off at 52.5 degree. When i took the iSpindel out of the water and suspended it vertically, the tilt angle was 42 degree, which I assume it should be near 0 degree when vertical. Tried re-setting, re-flashing, re-calibrating the offset. No improvement

 Would it be safe to say that the Gyro is a shot duck!

thegreatgunbantoad, 2020-07-13:
Almost certainly.  I've seen similar.  The gyros are very prone to being naff.

worgz01, 2020-07-14:
OK, changed the Gyro. 
I then went into configuration mode to do offset calibration. The angle was reading 86.4 degrees, so OK I'll have to offset cal.
I thought,  I'll see what the angle was in water (SG 0.000) before offsetting. Reading was 22.7. 
Excellent I thought, if the offset calibration moves to the 89.5 degrees, the angle at 0.000 gravity should be very close to 25 degrees. OK
Carried out the offset, checked the 0.000 gravity and low and behold, I'm back to the same issue. :-(

When I grab the device (out of water) and rotate it from horizontal through a rotation of approximately true 120 degrees, then I get readings of 25 Degrees.

Surely this is something other then the Gyro????

sirtomwwatts, 2020-07-15:
I was about not post an almost identical issue. Slightly different numbers but basically I can't get my gyro to go close to vertical
[image]
[image]
[image]

I'm using firmware 6.2.0, I don't have another gyro so would like to rule out hardware before I buy another

ErikdBr, 2020-07-15:
Did you try laying it flat on a table exactly as you did doing the offset calibration but then roll it over 180 degrees and then do the offset calibration?
I read somewhere on here that some of the builds, and there are several different boards with and without battery holder and the like, the gyro needs to be "upside down" when doing the offset calibration.

worgz01, 2020-07-16:
Yes, followed the directions as stated in #340 with no luck at all. Have done a factory reset, re-flash and offset (with the Gyro upside down) with no luck. Even tried with the GYRO in the vertical plan :-)
It just strange how it was close to the required angle of 25 degree in 0.000 SG before I did the offset calibration

thegreatgunbantoad, 2020-07-16:
Some of the gyros out there are just rubbish and only good for the bin.  Same can be said of the Wemos boards (most of which are clones).  Occasionally the temp sensors are duff too.  If you can, test everything before you solder i down.

closed 2020-09-22 by stale[bot]