universam1/iSpindel #404: Using large token like jwt

alejandroaap, 2020-07-18:
I recently had an iSpindel mounted and ran into a problem. I wanted to use a jwt token but it was impossible because trying to store the jwt token does not store it completely, being impossible with this firmware to use such long tokens. Can they update the firmware to use longer tokens? I am not an expert C or firmware developer so if anyone can help me I would appreciate it.

Regards.

ldab, 2020-07-30:
I believe you have better chances to find some help if you explain the reason for chosing jwt token as well as the benefits, etc.

alejandroaap, 2020-08-12:
Hi @ldab 
I need the ispindel to communicate with my server using a token that the server knows how to decrypt and obtain configuration parameters. I have increased the token size by code and this has solved my problem, now I have been able to add jwt tokens. It would be interesting to provide security options and for the token not to be in the body but in the header. I have not seen that there was an https library for sending post requests although I have to say that I am not an expert in C. Can you help me? https://github.com/universam1/iSpindel/issues/405

ldab, 2020-08-13:
https://github.com/universam1/iSpindel/issues/352

closed 2020-10-19 by stale[bot]