universam1/iSpindel #407: Ispindle almost impossible to find Wifi or get a logging

rfrins, 2020-07-24:
Hi,

I'm the owner of an Ispindel and it's almost impossible to find a WIFI signal and get a brewfather reading.
I have a dedicated WIFI router setup for the Ispindel.
It's a 2.4 ghz network only setup for the Ispindel. The Ispindel is very close to the router (for testing purpose)
For setting it up, I removed my Ispindel from deep sleep and I did the config via 192.168.4.1 via a laptop.
I can actually find my WIFI network, so I fill in the configuration for the WIFI network and for brewfather. 
Next my Ispindel connects once to brewfather and gives out 1 reading.
That’s it…. For the next 4-6 hours no reading is given to brewfather.
When I connect the Ispindel (by removing it from deep sleep) and I do a wifi scan I cannot find any WIFI networks any longer… none at all. (while looking it my phone there should be multiple available)
I’ve let it run for 24 hours now and was able to get 4 reading in those 24 hours…
It’s very strange, I would expect it either works or doesn’t, but not only once every 6 hours at a random point in time.

I’ve already tried the following as a solution:
On the Ispindel I’ve tried multiple different wifi signals. I’ve tried the dedicated signal from the router, I’ve tried the signal from my Orbi mesh network (mix of 2.4 ghz and 5 ghz), I’ve tried a wifi repeater and I’ve tried a hotspot from my phone. No result
On the dedicated router setup I’ve tried switching from an automated channel to a manual channel (tried manual channels 1, 6 and 11) and on the channel bandwidth i switched between 20mhz and 20/40 mhz

Is there a way to get into the logging of this device?
I would expect that if I plug into a laptop USB I could see it in my explorer, but this is not the case. I cannot find an additional USB drive when connecting it.

Any tips or tricks from people wo have experiences such issue?

mp-se, 2020-07-24:
You can do logging via an usb cable and debugger. I use vscode with platformIO for that but there are more debuggers for audrino available. I guess any serial console with the correct baud rate should work. You will need to install a driver for usb-serial for this to work. 

This is the easiest way to check whats going on and get others to help. 

First, are you sure its waking up more than once? It should activate the blue led while active. If you have the wrong resistor connected to the rst pin it might not wake up again.

rfrins, 2020-07-25:
Thank you for your response.
I'm sure everything is connected correctly. I get a reading every 6-12 hours in Brewfather

I'm going to try and read the logging.
I'll get back when I succeed with this

davidfascio, 2020-07-28:
Hi, before continue discussion. Have you tried 900 seconds (15 minutes) as period reading time ? From brewfather docs https://docs.brewfather.app/integrations/ispindel, it says any smaller value than 900s will be ignored, and no readings are going being log.

Regards.

thegreatgunbantoad, 2020-07-28:
To Log you need the USB port on the Wemos board (probably actually a clone board) if you have a PCB variant that USB will be sort of half way down.  You should be able to connect via putty of hyper term (I too tend to use VS code and platform IO though).  Check the orientation of the diode, if that's backwards you'll never get out of deep sleep in normal running.  In all likelihood it's a duff Wemos clone board that needs replacing.

rfrins, 2020-07-29:
Hi,

Yes this is my setting indeed !

Br

projects-in-progress, 2020-09-28:
I have a similar issue see #424

closed 2020-12-04 by stale[bot]