universam1/iSpindel #408: Diode to Protect 18650 from Over-Voltage Damage

OpenSourceDistilling, 2020-08-03:
Hello, 

A colleague suggested that I add a diode to The Jeffrey PCB to protect the 18650 lithium-ion battery from the 5 volts coming from the D1 mini when the D1 mini is plugged in via its USB port. 

Right now I'm suggesting that everyone remove the diode from the D1 mini to protect the battery from the 5 volts coming from the USB port. Once the diode is removed from the D1 mini the D1 mini can no longer be powered from the USB port but instead needs to be powered via the battery. This leads to all sorts of confusion as many people building iSpindels aren't familiar with what a diode is or how it works. On top of that, removing the diode from the D1 mini means it can no longer be flashed like an out of the box D1 mini, leading to further confusion. 

I don't know very much about selecting a diode for this application or if it will work for this particular application. I'm hoping someone smarter than me can assist with the below questions: 

1. Is the theory sound? Can we simply add a diode as pictured in the schematic below (or somewhere else) to protect the lithium battery without having to modify the D1 mini directly?
2. If we can add a diode to protect the battery what model of diode should be used? I would be looking at adding a through-hole mounted diode to my PCB design. 

[image]

ldab, 2020-08-05:
Do you have the full schematic as well as the TP-4056 datasheet, because the one I found it's completely different: https://datasheet.lcsc.com/szlcsc/1912111437_TPOWER-TP4056_C382139.pdf

OpenSourceDistilling, 2020-08-05:
Hey @ldab, thanks for the reply. The above schematic is actually referencing the TP4056 breakout board and not the actual TP4056, sorry I should have been more clear about that. Here is a pic below of the "TP4056" which is actually the TP4056 breakout board that I'm using for my builds. 
[image]
 
Here is the full schematic for my PCB: https://www.opensourcedistilling.com/wp-content/uploads/2020/03/Printing-Print-Schematic-scaled.jpg

This issue is that the 5 volt in on the D1 mini becomes a 5 volt out when the D1 mini is connected via the USB port, which means it's sending 5 volts to OUT+ on the TP4056 breakout board. I'm hoping that we can add a diode to protect the battery when this happens.

ldab, 2020-08-05:
If this TP4056 board have the schematic [1], you will notice that B+ and OUT+ are shorted and as mentioned, when applying VCC to the wemos, one will apply the same voltage to the battery, which is not recommended at all, since it's higher than most batteries Vmax.

DW01A [2] is a battery protection IC, by connecting OUT- to IN- you disable this IC.

To answer your question, yes a diode can be added, pretty much any diode would work with the cathode connected to TP4056 OUT+, just need to make sure that the diode can handle the current supplied by the battery ~500mA. But this will add some significant loss. A more sophisticated option would be to add a PMOS.

I would recomend you having a look at D1 mini Pro which cover the topics mentioned above [3]

[1]: https://www.sunrom.com/media/content/1203/DW01A.gif
[2]: http://hmsemi.com/downfile/DW01A.PDF
[3]: https://docs.wemos.cc/en/latest/d1/d1_mini_pro.html

OpenSourceDistilling, 2020-08-05:
Hey @ldab, on the TP4056 breakout board OUT- to IN- are connected. You can see this on the schematic you provided ([1](https://www.sunrom.com/media/content/1203/DW01A.gif)) and I tested them with a multimeter just to be sure. The connections on my PCB are redundant. Are you saying that the DW01A would protect the battery from 5 volts coming in on OUT+ but the DW01A is disabled on the breakout board out of the box? My assumption is that the 5 volts will go directly into the battery from OUT+ regardless if the DW01A is there are not. The D1 mini and the TP4056 breakout board share a common 'ground'.

You mentioned that the current supplied to by the battery would be ~500mA. Would the BAT43 suffice? You mention "significant loss", do you mean that the diode will consume a large amount of energy significantly reducing the iSpindel's battery life? 

I looked at the D1 mini Pro some time ago but wifi antenna and USB charge ports are on opposite sides. Aside from having half the amperage for charging when compared to the TP4056, the D1 mini Pro would need to be installed upsidedown for the charging port to be accessible via the lid of the iSpindel resulting in the wifi antenna being further into the fermenting liquid. Wifi signals and water do not mix and would assume this would have a significant impact on the iSpindel's wifi range. If the D1 mini Pro was used the entire assembly would need to be removed from the petling each time the iSpindel needed a charge. I am under the impression that removing the electronics from the petling after assembly should be done only for maintenance reasons like replacing the battery or replacing components. Since the gyro is so sensitive I think constantly removing the elctronics could have a negative effect. What do you think, am I paranoid?

ldab, 2020-08-05:
> TP4056 breakout board OUT- to IN- are connected

My bad, got the labels mixed up.

> DW01A would protect the battery from 5 volts coming in on OUT+

Potentially, as when the DW01A sense over-voltage it should open the Mosfet, disconnecting the battery. The datasheet is a bit poor and I'm not familiar with this IC and would prefer to not comment further.

> Would the BAT43 suffice?

Potentially yes, [datasheet](https://www.vishay.com/docs/85660/bat42.pdf) indicates continuos Forward Current of 200mA and peak 500mA. I would recommend against it, unless you're 100% sure that the values are within specification.

> You mention "significant loss", do you mean that the diode will consume a large amount of energy significantly reducing the iSpindel's battery life?

If we take BAT43 as an example, when IF = 200mA, the forward voltage is 1 Volt, so if we assume a full 4.2 battery, it will only deliver 3.2 Volts to the Wemos.

> What do you think, am I paranoid?

Without further testing side by side is hard to compare, but I have been using mine like the way you mentioned, upside down, inside the liquid, etc.. The Fermenter is inside a cabinet and 15m from the router, and I'm happy with the RSSI (-80dBm).

[image]

OpenSourceDistilling, 2020-08-05:
@ldab Thanks man! You've given me lots to think about. I will ponder this for a while and do some research. Maybe a 3 position switch would be a different way to approach a solution to the problem. One position each for: off, on, and D1 mini flash mode. Thanks again for your help I really appreciate it!

Cheers!

Joey Joe Joe Jr.

lbussy, 2020-08-05:
We had talked about the use of a gravity switch for things like this, or for completely shutting the controller down when upside-down.  It's an option.

I like the PMOS idea, or using the Pro.  I have no idea if it would fit, but it's a nice board.

OpenSourceDistilling, 2020-08-06:
I just bought 2 of the D1 mini Pros to play around with. We will see how long they take to get here from China. Hopefully, they will fit. Should be fun to experiment with them. I think I need a bigger box for all the electronics I keep accumulating, ha ha. 

I will look into PMOS more. From what I gather you can set a pin on the PMOS to high and get a low output, or set the pin on the PMOS to low to get a high output. I'm new to building electrical circuits. Do we have any ideas about what mechanism from the iSpindel we would use to control the PMOS? If you have an idea can you please sketch out a schematic on a napkin? I'm drawing blanks... When USB power is applied to the D1 mini current would be flowing toward OUT+ instead of away from it, would that be the trigger somehow? 

I did some more digging with a D1 mini, a multimeter, and a power supply. 

When the D1 mini is plugged into the USB, the 3.3-volt pin =~3.3-volts and the 5-volt pin =~4.5-volts. 
When the D1 mini is not plugged into the USB and is instead powered by the 5-volt pin with ~5-volts, the 3.3-volt pin is equal to ~4.5-volts. 

So there is some sort of detectable difference between the two different ways of powering the D1 mini v3. Is that information any use to us for controlling a PMOS?

Thanks for your assistance. There are so many things to learn.

ldab, 2020-08-06:
> I'm new to building electrical circuits.

There's a lot to be learned on the electronics world. As mentioned previously, check the wemos D1 mini pro schematic, it implements it already, check section "Power". When the USB is connected, voltage is applied to the gate, disconnecting the battery, making the current flow thru D2.

[image]

> I think I need a bigger box for all the electronics I keep accumulating

It does not take a lot of space, check below.

[image]

OpenSourceDistilling, 2020-08-06:
@ldab Fantastic! Thank you! You have been much and I appreciate your assistance. Once again, many thanks!

allthepies, 2020-08-06:
Hi, 
I am "the colleague" in question 👍 
Good discussion, seems it's not quite as simple as just adding a diode.

I've got a couple of D1 Mini Pros and have some external antenna coming with the intention of building up an iSpindel with that option.
I see the above ^^^ Mini Pro is the "new" version, the models I have are the same form factor as the standard D1 Mini and will fit into the Jeffrey PCB as-is.

OpenSourceDistilling, 2020-08-06:
> I've got a couple of D1 Mini Pros and have some external antenna

Are you speaking about one of these guys? 
[image]

I have a few of those kicking around as well. They are a regular D1 min but with a ceramic antenna with the ability to add an external antenna. There is no onboard charging with these (just like the D1 mini v3), so we will run into the same battery over-voltage issue. 

After ponding PMOS further, I think it's only possible if we solder a lead to the 5 volts coming from the USB port on the D1 mini v3 but before the diode on the D1 mini (like what's pictured in the D1 mini Pro). Between these two components is the only place that will be energized with USB power only. I think this is the only way to definitively discriminate between power coming from the USB and power coming from the voltage regulator on the D1 mini. But if we do such a thing we are back to modifying the D1 mini directly which is what I wanted to get away from in the first place. 

Are there any patent lawyers out there (LOL)? How many components can I integration into a single PCB before I lose my house? It would be nice to have a way to turn off power to the CH340 when the firmware isn't being flashed via USB port and I know that CherryPhilip created a PCB with the gyro integrated. I found a single board iSpindel design online somewhere with a small form factor. I see a company selling iSpindel kits in Europe. With a single PCB with all components, we could probably get the iSpindel small enough to fit through the hole on a standard glass fermenter. But now I'm just thinking out load.

beikeland, 2020-08-06:
@OpenSourceDistilling  Patents lawyers will have food for anything that uses the same principle of measuring regardless of construction afaik. You see kits in the EU because the patent is only valid in the US, you can build whatever you like for your own non-commercial use (or maybe the wording is educational or something) without stepping on patent toes - but you cannot sell it for sure.
I was after getting one down the neck of a 1 gallon glass carboy but finding a combination of a small enough container and a battery(weight) that gives a good angle when floating while keeping the run-time reasonable seems a bit of a task..

ldab, 2020-08-06:
https://patents.google.com/patent/US20140260607A1/en

OpenSourceDistilling, 2020-08-06:
> You see kits in the EU because the patent is only valid in the US

[He seems to have applied on WIPO](https://patentscope.wipo.int/search/en/detail.jsf?docId=US107404595&tab=NATIONALBIBLIO&_cid=P21-KDJEZY-46164-1), meaning that the patent is enforceable in 153 countries. At least that's how I understand this entry in the WIPO database. Most countries have already signed a patent treaty meaning if it's on WIPO the border doesn't really mean anything, the patent is enforceable in other countries. But I'm no lawyer. I looked into this some time ago just out of interest. 

I guess my question is if I created a single PCB with all components except the battery holder (without a battery it doesn't really do anything right?), and PCBWay (or other PCB printing company) created it and sold it, would I somehow be liable for designing it, or would the manufacturer be liable, or would no one be liable because it's not fully assembled and not a functional piece of equipment when it's shipped? I don't know any patent lawyers and certainly won't be shelling out money to find out the answer. 

For example, you can buy all the components to build a taser. If there is no battery holder and no battery is it really a taser? Or is it just a bunch of junk thrown together that doesn't do anything? Interesting thought experiment but I have no idea what the legal answer is.

OpenSourceDistilling, 2020-08-06:
Coming back to the idea of a diode.

If we use a diode we get a voltage drop from the battery. If a fully charged battery is about 4.3 volts. [Wiki says ](https://en.wikipedia.org/wiki/Diode#:~:text=In%20a%20small%20silicon%20diode,V%20and%204.0%20V%20respectively.)we can get about a 0.7 volt drop with a small silicone diode. 4.3-0.7=3.6 volts

[According to this website](https://diyi0t.com/what-is-the-esp8266-pinout-for-different-boards/), we can power the D1 mini via the 3.3-volt pin with 2.5 to 3.6 volts. I wonder if this is a viable option and what the actual voltage threshold is that will cause the D1 mini to power down when it's being powered via the 3.3 volt pin. 

This same site says a minimum of 4 volts when powering the D1 mini via the 5 volt pin but from what I've seen the D1 mini will keep going until around 3 volts on the 5 volt pin, then it will shut down.

Is the theory sound and does it warrant further investigation? 

https://diyi0t.com/what-is-the-esp8266-pinout-for-different-boards/
[image]

allthepies, 2020-08-07:
> Are you speaking about one of these guys?
> [image]
> 

Yup, that's a D1 Mini Pro (earlier model)

ldab, 2020-08-07:
@OpenSourceDistilling you're missing some basic concepts on your question.

> 0.7 volt drop with a small silicone diode.
 
The voltage drop is not linear and will change accordingly to the current, as shown previously. Any diode datasheet will have this information. you can also use a schottky diode.

> D1 mini via the 3.3-volt pin with 2.5 to 3.6 volts

ESP8266 [datasheet](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf) confirms the voltage level, however the battery voltage fluctuates as well as the forward voltage on the diode, meaning that when the module connects to WiFi and transmit data, the current will increase (500mA) and the voltage if floating around 2.5V will drop, guaranteed, making the module reboot or do weird stuff.

> going until around 3 volts on the 5 volt pin, then it will shut down.

Yes, but again, if you look at ME6211 [datasheet](https://datasheet.lcsc.com/szlcsc/Nanjing-Micro-One-Elec-ME6211C33M5G-N_C82942.pdf) you will notice that the drop out voltage is 0.5V, this number is the minimum required to provide output current of say 400mA, i.e 3.3 + 0.5 = 3.8V min on the input voltage. The module may still work until 3V but the regulator will not be able to supply the required current, you're now working an unkown, not guaranteed region.

allthepies, 2020-08-07:
> [He seems to have applied on WIPO](https://patentscope.wipo.int/search/en/detail.jsf?docId=US107404595&tab=NATIONALBIBLIO&_cid=P21-KDJEZY-46164-1), meaning that the patent is enforceable in 153 [...]

I'm also no lawyer but his patent has a "US" prefix which I believe means it's applicable in the US only.  I'm in UK 👍

OpenSourceDistilling, 2020-08-07:
> I'm also no lawyer but his patent has a "US" prefix which I believe means it's applicable in the US only. I'm in UK

I would have to ask why is his patent in the World Intellectual Property Organizations database if his patent is only applicable to the US? I believe the reference number starting with US is just the original US patent number for lookup purposes.

But hey, sounds like you got it all figured out. You should start an iSpindel manufacturing business. You could seriously make a ton of money 👍

The truth is I'll be sticking to making just PCBs for now until I meet a lawyer friend that can calm my nerves.

That being said, sometimes it's easier to ask for forgiveness instead of permission. 

Anyways, we are getting off-topic, lol. There will only be speculation and endless debate around a subject like this without a professional involved as none of us actually know what we are talking about, **_me included_**.

allthepies, 2020-08-07:
> But hey, sounds like you got it all figured out. You should start an iSpindel manufacturing business. You could seriously make a ton of money 👍

I do have a few PCBs here :)

OpenSourceDistilling, 2020-08-07:
@ldab Thanks again for your response. I will educate myself more about didoes and see if I can make better sense of things. I did a few tests with my multimeter and a BAT43 last night and measured a 0.35 voltage drop with 5 volts power supply at 100mA. More reading and experimentation is definitely required. Apparently Schottky diodes are known for their low voltage drops. I also read that the voltage will creep back up when a device is sleeping and then drop back down when it draws power after waking up, further complicating things. Again, so many things to learn.

In regards to ME6211, from what I can see on the [schematics ](https://escapequotes.net/wp-content/uploads/2016/02/2020-02-10_12-06-36.jpg)the 3.3 volt pin powers the EP8266 directly and bypasses the ME6211. From what I can tell the ME6211 is used as a step-down voltage regulator on the D1 mini and doesn't really come into play when using the 3.3 volt pin. 

Thanks again for all your comments. You're really helping me get things into perspective. About 20 years ago in college, I knew circuits pretty well but now that seems like a past life.

Be well and be safe!

OpenSourceDistilling, 2020-08-07:
> I do have a few PCBs here :)

He he he, I won't tell anyone ;)

beikeland, 2020-08-07:
> [He seems to have applied on WIPO](https://patentscope.wipo.int/search/en/detail.jsf?docId=US107404595&tab=NATIONALBIBLIO&_cid=P21-KDJEZY-46164-1), meaning that the patent is enforceable in 153 countries. At least that's how I understand this entry in the WIPO database

You could just ask Baron Brew, i did and they confirmed only having a US patent back in January 2020. Presumably they should know, without being lawyers.

If you're looking for a loophole, maybe loading a firmware that does something different, letting the end user obtain and flash the iSpindel firmware themselves?

OpenSourceDistilling, 2020-08-09:
> You could just ask Baron Brew

What a great idea @beikeland! I tracked down their email address and just sent an email off asking if they are able to clarify if their patents are US only or International. I will update this thread when they reply. It would be nice to be able to build a single PCB with all iSpindel components already on it, it would then be easy to fix the battery over voltage issue.

OpenSourceDistilling, 2020-08-12:
Baron Brew got back to me. He didn't specifically say the patent wasn't international, but he did say that the patent was active in the US. He also said that the software has a Copywrite and that "Tilt" is trademarked. 

I read up more about the patent process. From what I can tell **my initial thoughts were wrong**. WIPO doesn't automatically grant patents in 153 countries but instead provides a 3 year grace period (or maybe less time) to file your patents in other countries while having those countries retroactively issue your patent on the date of the original patent. So if your US invention was patented Jan 1st, 2020, and two years later you issued the same patent in Japan the Japanese patent would also be effective Jan 1st, 2020. You have to pay lawyer fees for each country you want to get your invention patented in (unless you do it yourself). A US patent can cost tens of thousands of dollars so patenting in multiple countries will add up very quickly $$$ and you would need a business reason to do so. I also believe there is an international patent application format where the single patent can be submitted to multiple countries using a single standardized form.

The Tilt patent was issued in 2014, the 3 year grace period has passed. I don't live in the US and I checked my country's patent database where I was unable to find the patent for the Tilt. So I believe this means that I can create a single integrated PCB with all the required iSpindel components without fear so long as the finished products don't get imported into the US. I believe I would have to order and sell the PCBs directly to actively keep them from entering the US and therefore I don't think that particular design could be open source :( If a product I designed started showing up in US homebrew shops I could get in trouble. 

I can see that the European seller of iSpindel kits no longer ships to the US. I assume this is for legal reasons...

I do have a few lawyer friends none of which are involved in patent law. I did ask one tonight about my understanding of patents and he said with his 'limited knowledge' he understood that same to be true. He said if the products never enter the US I should be fine unless someone else issues a patent in the country I'm in, at which point I would need to stop production.

Given the scope of personal involvement in a project like this, I'm unsure if I really want to proceed. I think I would need to facilitate the shipping which means storing and packaging many units and individually shipping them out. It would be more like a business thing than a hobby thing. I already have a full-time job. 

What an interesting journey from posting about a diode to learning about patents. With building my own PCB I could definitely solve the over-voltage issue. I guess I will see if I can muster up the motivation to make this sort of thing a reality. 

Thank you to everyone for your help and comments. Very much appreciated! It's been eye-opening.

closed 2020-08-12 by OpenSourceDistilling

beikeland, 2020-08-12:
> If a product I designed started showing up in US homebrew shops I could get in trouble.

it would generally be the importer facing infringement issues, when the product is made and sold outside the US.

in any case, the reasonable thing to do would have been to recognize the effort from both the Tilt and iSpindel and get the appropriate authorizations (or attempt to at least)

darkporpoise, 2020-09-02:
> There's a lot to be learned on the electronics world. As mentioned previously, check the wemos D1 mini pro schematic, it implements it already, check section "Power". When the USB is connected, [...]

@ldab  that looks awesome, I've just bought a [DI Mini Pro V2.0.0](https://www.aliexpress.com/item/32724692514.html) and was hoping to use it for an iSpindel. Do you have a wiring diagram for your setup?

ldab, 2020-09-03:
it's the same schematic posted on the [instructions](http://www.ispindel.de/pics/schematic-en.png). the only difference is that i set pin `D5` to GND so that i don't need a jumper.

darkporpoise, 2020-09-03:
Thanks ldab. I was wondering what it looked like without the TP4056 as the Pro seems to have one built in. I think I can work it out from those instructions though (electronics is not my strongest suit).

ldab, 2020-09-03:
Yes, just do not use TP4056.

OpenSourceDistilling, 2020-09-04:
> the only difference is that i set pin `D5` to GND

Thanks for sharing. I can see how that allows the gyro to be mounted with pins and no jumper, D5 isn't used by the iSpindel. I'm planning on building a pro version 2 like yours on a breadboard this weekend to see if I can get it working, then move into a petling. The pro v 2 is surprisingly small and I have received the 2 I order some time ago. I also plan on building an iSpindel with a ceramic antenna this week as well. I am interested to see how the different wifi signal strengths pan out between the D1 v3, D1 with ceramic antenna, D1 mini pro v2. I wish they named these things better.

darkporpoise, 2020-10-28:
> I'm planning on building a pro version 2 like yours on a breadboard this weekend

How'd it go (if it went at all)?

OpenSourceDistilling, 2020-10-30:
Hey @darkporpoise I ended up getting the D1 mini pro (the green one) and had initially planned on using it for my next PCB design. I then came to the conclusion that the D1 minis are the widest part of the current iSpindel. I figure that the only way to fit them into smaller tubes is to use something like an ESP12 instead. The current tubes are not that easy to get in the Americas and covid has only made that worse. Anyways, long story short I'm now waiting on an ESP12 programmer to flash the iSpindel firmware and build that on a breadboard. So I kind of dropped the ball on the whole D1 mini pro thing. I want to make a smaller version of the iSpindel that will fit into smaller tubes, hopefully, a wide variety of tubes instead of just the one type that most people currently use and are commonly sourced from Europe.