universam1/iSpindel #411: New features

alejandroaap, 2020-08-12:
I have three questions that I would be delighted to have someone answer.

Would it be possible to access the ispindel without having to put it in configuration mode? The purpose is to be able to connect to the ispindel through a mobile application and modify its settings. One way I can think of would be that if it is not connected to a Wi-Fi network, the ispindel automatically goes into configuration mode.

Is it possible to request a username and password to access the ispindel configuration?

Would it be possible to update the firmware via OTTA?

Thank you

ErikdBr, 2020-08-12:
One question I can answer, updating the firmware OTA is allready possible.

alejandroaap, 2020-08-12:
@ErikdBr Thanks for answering. You can provide me information about how to update the firmware via OTA

allthepies, 2020-08-12:
> @ErikdBr Thanks for answering. You can provide me information about how to update the firmware via OTA

It's in the documentation

https://github.com/universam1/iSpindel/blob/master/docs/Firmware_en.md

alejandroaap, 2020-08-12:
@ErikdBr I had already read the documentation and I have tried to update the ispindels from the configuration mode simulating the process that is done from the web. What I want is to send an update to the ispindels without having to put them in configuration mode, in the same way that we can update our smartphones without a person coming to update them. Thanks you very much for your help.

ErikdBr, 2020-08-12:
I wasn't the one answering about the documentation thans you.
However, what you want would be nice but is not possible at the moment. I think that needs a major rebuild of the firmware if at all possible with this hardware.

alejandroaap, 2020-08-12:
@ErikdBr Sorry, I did not look at who had answered me. Do you know any project on which I can base myself to try to implement what I am saying? I am not specialized in this type of programming but I do not want to miss the opportunity to try it since right now it is one of the handicaps that I see, since it would not be to put 1 device into production but it would be to have between 100 and 200 and although firmware versions are not going to be released every day I would like to be able to offer that possibility.

ldab, 2020-08-13:
> Would it be possible to access the ispindel without having to put it in configuration mode?

No, the developer is not in favour with that -> https://github.com/universam1/iSpindel/pull/264#discussion_r374052346

> Is it possible to request a username and password to access the ispindel configuration?

Yes, add as the second parameter here: https://github.com/universam1/iSpindel/blob/ff154a99d2a8b53395965c9c7661a7118ea31b05/pio/src/iSpindel.cpp#L398

> between 100 and 200

It sounds like you're trying to do more than a DIY OpenSource which is the aim of this project, this can be achieved by having your device to check Updates via HTTP to your server, check https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#http-server It will take a lot of memory, not sure if

closed 2020-10-19 by stale[bot]