universam1/iSpindel #413: iSpindel AP not visible (or not created), with arduino debug log

camjose, 2020-08-16:
Hi everybody, 

I am struggling with my brand new, home made, iSpindel. I can not go into the config mode because I don't see any wifi SSID named iSpindel AP or something. 
[image]

Flashing the firmware seemed OK using NODEMCU, 115200 baudrate etc... everything as recommended in the FAQ. I tried FW5.8.6 as recommended in #324 I also flashed 6.2.0 and 6.3.0, but no improvement. For sure, I did initialize every time I change firmware as recommended in [here](https://github.com/universam1/iSpindel/issues/275#issuecomment-510397706) 

Contrarilly to issue #364 or #221 , the temperature sensor seems OK, indicating 30°C or so...
I did debugging with arduino monitor :
The iSpindel goes into config mode, starts the http server ...  I guess that at this point I am supposed to see a wifi network nam iSpindel pop upon my phone or laptop.. ?? Well, I don't.

After few minutes, new lines appear on the log : 

"After waiting...
*WM: 0.00
*WM: seconds
*WM: Timed out connection result: 
*WM: WL_IDLE_STATUS

Here below is the complete debug log with arduino : 

[image]
[image]
[image]

Any help would be appreciated.... I think I've read each one of the issues reported on Git...
Thank you
Cam

allthepies, 2020-08-16:
Perhaps you have a faulty Wemos ?   I had one once which seemed fine but the Wifi was inoperative.
Perhaps flash a sketch which just reports the Wifi networks it can find locally ?

camjose, 2020-08-17:
Thanks for your rply, that sounds worth to try.
I have every component spare so I will mount a 2nd circuit and see if it behave the same.

ldab, 2020-08-18:
Maybe you choose the wrong Flash Size? What tool did you use to flash the firmware?

camjose, 2020-08-20:
@ldab I used NODEMCU with the settings given in the wiki.
[image]

chesterweirdo, 2020-08-20:
> Perhaps you have a faulty Wemos ? I had one once which seemed fine but the Wifi was inoperative.
> Perhaps flash a sketch which just reports the Wifi networks it can find locally ?

How would you go about doing this?

chesterweirdo, 2020-08-27:
Have you managed to get any further?

I want to try and flash a simple firmware that acts as a web server just so that I can check it’s not the board. However I do not know how. Can anyone advise?

ldab, 2020-08-27:
Can you both post a picture or specifically tell which board are you using

chesterweirdo, 2020-08-27:
[image]
This is the one I’m using.

camjose, 2020-08-28:
I have got the exact same Wemos d1m1n1

ldab, 2020-08-28:
This has a 4MB flash which dos not support my initial suggestion.

I would try to flash another firmware to see if works, as a suggestion you can try "tasmota" which has the same-ish access point mechanism, the binary is here: https://github.com/arendst/Tasmota/releases/download/v8.4.0/tasmota.bin

chesterweirdo, 2020-08-28:
I’ll give that a try a bit later. 

I did try the 6.0 firmware for the ispindel but got the same issue.

chesterweirdo, 2020-08-28:
Okay so I tried the Tasmota firmware. Interestingly it showed as an AP on my Mac but would not let me join it. Thats better than what I got with the iSpindel.

chesterweirdo, 2020-08-28:
Okay so something is strange here. I now have flashed the iSpindel firmware and on my iMac I can see a wifi network called iSpindel. However, my Mac book can not see it or my phones. On the iMac that can see it it will not allow me to connect.

I then git this in the terminal

`started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: After waiting...
*WM: 0.00
*WM: seconds
*WM: Timed out connection result:
*WM: WL_IDLE_STATUS

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40068870 depc=0x00000000

ctx: cont
sp: 3ffffa00 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffba0:  ffffffff 3fff255c 00000000 00000000
3ffffbb0:  3ffeffb8 402156e3 00000055 3ffe920d
3ffffbc0:  00000010 3ffffc20 00000000 3ffe9eeb
3ffffbd0:  3ffeffb8 0000039b 0000039b 00000000`

ldab, 2020-08-31:
How are you powering the board?

camjose, 2020-09-01:
At the moment I am using a stabilized power supplier, set at 3.7 V. It is connected to the "battery  IN" connectors.
 I wanted to test my circuit before to plug the 3.7 li ion battery.
Could the use of a power supplier instead of the battery really be the issue??

ldab, 2020-09-01:
I think there is your problem. Do you mean B- and B+ as per [this](http://www.ispindel.de/pics/schematic-en.png) diagram?

If positive, the wemos LDO ([ME6211](https://datasheet.lcsc.com/szlcsc/Nanjing-Micro-One-Elec-ME6211C33M5G-N_C82942.pdf)) has a dropout of 680mV@500mA. So I would say that when the WiFi kicks in and draws 500mA the voltage drops to (3.7 - 0.68) 3V potentially causing some issues, can you please try with the USB power or increase your power supply to 4.1V?

chesterweirdo, 2020-09-01:
Mine is running off the battery. It was fully charged but I could try charging again.

camjose, 2020-09-02:
> I think there is your problem. Do you mean B- and B+ as per [this]

Yes, B- and B+. 

> If positive, the wemos LDO ([ME6211](https://datasheet.lcsc.com/szlcsc/Nanjing-Micro-One-Elec-ME6211C33M5G-N_C82942.pdf)) has a dropout of 680mV@500mA. So I would say that when the WiFi kicks in and [...]

I'll try this next sunday, i'll be home. 
Thank you

ldab, 2020-09-07:
It could be a problem, but when the board is on Accesspoint it may drain more current (i don't know). and the li-ion battery, when charged is around 4.2v

camjose, 2020-09-07:
Hi, 
@ldab , I have tried with 4.3 V on B- B+ using my stabilized power supply, unfortunately the wifi AP is still not visible.
@ldab too, I have tried using Tasmota firmaware. No wifi AP visible. See below the log : 
[image]

All this suggest a hardware issue, so next step will be to build a new board. This will be done this week I hope.
Thank you for your support
CJ

camjose, 2020-09-19:
Hi, 
I've juste finallized the soldering and testing of a new board.
Everything is new on this one, new wemos, new sensor, new gyroscope, new everything....
Unfortunately the results is exactly the same.
No ispindle AP visible, nor Tasmota when I try with that other firmaware.

Is my soldering correct.... I'll post pictures

chesterweirdo, 2020-09-19:
That’s a shame that the hardware has not solved it. I had not bought a new d1 yet and I have to admit I was holding off as I can not think that many are faulty. It does feel like something else.

thehammertownhead, 2020-09-21:
I was having similar issues.

one unit flashed fine and worked without any problems and the second unit did not.

Looking in the serial log I saw it failing ... fails mounting the file system and at some point after reading the temperature, it is giving temperature sensor errors......... then it completely errors out and reboots similar to above

when I look at the AP on my phone and try to connect it just disconnects.....likely when its rebooting...

I wrote some custom code to test the Filesystem reading and writing and that worked fine, I did the same thing for temperature sensor and that worked fine too, I did the same thing by adding some more code to test the Gyro and it worked just fine too. So my hardware and soldering is perfectly fine.

I had to change some code in V6.3.1 and hardcode my credentials to write them to the FileSystem on startup and then was able to access the AP. Its a pain to setup Arduino to compile it because you need to install all the supporting libraries......but its nice to have it working. The problem must be related to not having a config file on startup.....but I don't know exactly why there is an issue, so this hack gets around the problem.

here is the code I modified...

```
void setup()
{

  Serial.begin(115200);

  CONSOLELN(F("\nFW " FIRMWAREVERSION));
  CONSOLELN(ESP.getSdkVersion());

  sleepManager();

  bool validConf = readConfig();
  if (!validConf){
    CONSOLELN(F("\nERROR config does not exist or is corrupted"));
    strcpy(my_name, "iSpindel0002");
    strcpy(my_token, "AddyourUbidotsTokenHere");
    my_OWpin = 12; //If you know the ESP pin for your temp sensor you can set it here otherwise comment it out.
    my_ssid = (const char *)"AddyourhomenetworkSSIDhere";
    my_psk = (const char *)"Addyourhomenetworkpasskeyhere";
    my_api = 0; //for ubidots
    //my_api = 2; //CraftBeerPi
    //my_api = 3; //HTTP
    //my_api = 4; //TControl
    //my_api = 5; //FHEM
    //my_api = 6; //TCP
    //my_api = 7; //iSpindel.de
    //my_api = 8; //InfluxDB
    //my_api = 9; //Prometheus
    //my_api = 10; //MQTT
    //my_api = 11; //ThingSpeak
    //my_api = 12; //Blynk
    //strcpy(my_server,"");
    //strcpy(my_port,"");
    //strcpy(my_uri,"");
    //strcpy(my_vfact,); //Battery conversion factor
    formatSpiffs();
    Serial.println("(setup) Mounting FS...");
    
    if (!SPIFFS.begin()) {
    Serial.println("(setup) Failed to mount file system");
    return;
    }
    else{
      if(saveConfig()){
        Serial.println("(setup) Config saved to SPIDFF.");
        validConf = readConfig();
      }
    }
  }
 
  if (!validConf)
    CONSOLELN(F("\nERROR config corrupted"));
  initDS18B20();
  initAccel();
```

thehammertownhead, 2020-09-21:
These are what my error logs looked like before I changed that code.......

[image]
[image]

allthepies, 2020-09-21:
If you use PlatformIO to build the project, rather than Arduino, then it's a click the button to compile job.  All libraries get installed for you.

thehammertownhead, 2020-09-21:
ah ok, never used PlatformIO before....will have to give it a try.

is it a completely different compiler than the arduino compiler?  maybe its a compiler issue?

allthepies, 2020-09-21:
> ah ok, never used PlatformIO before....will have to give it a try.
> 
> is it a completely different compiler than the arduino compiler? maybe its a compiler issue?

Probably the same compiler, just a different IDE.

chesterweirdo, 2020-10-06:
@thehammertownhead could you explain how you changed the code. Willing to give this a try but unsure what file to change. Also then how to compile it. Could you give me some pointers?

thehammertownhead, 2020-10-06:
I guess it depends how you want to compile it. I used Arduino.... which maybe isn't as straight forward as PlatformIO.

1. Install Arudino
2. create a new sketch
3. copy the code from https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp
to you new sketch.
4. save your sketch
5. copy all the library libe files from https://github.com/universam1/iSpindel/tree/master/pio/lib
into the sketch directory
6. compile and look at the compiler error then add the needed library......repeat until it compiles
7. make the modification to the code as seen above, substitute into the code above as required for you setup.

its likely way easier in Platform IO instead of arduino, but I haven't tried it so not sure.

But those are the highlevel steps

camjose, 2020-10-27:
Hi, 
After instaling all the needed libraries, I am now testing the wemos itself using Arduino's ESP8266Wifi example named "WifiScan". I simply connect the board to the COM4, upload :
[image]
Then I open monitor, reset the board...
And Tadaaaaa !
[image]
No network found...  :-( Maybe I should add that I also tried with external power supply, in case the USB port isn't sufficient.
It seems that none of my 5 wemos 3.0.0 have a NOT working wifi signal. Could they be bad clones ? Can somebody copy a link where i can purchase official boards ?

Thank you
Cam

ErikdBr, 2020-12-26:
> Hi,
> After instaling all the needed libraries, I am now testing the wemos itself using Arduino's ESP8266Wifi example named "WifiScan". I simply connect the board to the COM4, upload :
> [image]
> Then I [...]

Many of the v3.0.0 are faulty clones. I use the Lolin, previously Wemos, v3.1.1, here yiu can buy the official ones: https://a.aliexpress.com/_BOjvTs

thehammertownhead, 2020-12-26:
I used the ones from this Seller. One worked and the other one required that hack....I described above...but they seem to work fine.

https://www.aliexpress.com/item/4001291931302.html

ErikdBr, 2020-12-27:
> I used the ones from this Seller. One worked and the other one required that hack....I described above...but they seem to work fine.
> 
> https://www.aliexpress.com/item/4001291931302.html

I bought these ones too, their sh*t imo. They should work without altering any code. I rather pay double the price and have no hassel. Nothing is more anoying then soldering everything together and then find out you did it all for nothing.

camjose, 2021-01-03:
Hello, 
Happy new year to all of you.
I finaly received another board from one other seller. This time it works fine. I have also noticed that this last has a "wemos.cc" written on it, where the others don't...
So I believe they are poor quality copies...

closed 2021-01-11 by camjose