universam1/iSpindel #414: Flashed iSpindle but do not see Wifi to setup.

chesterweirdo, 2020-08-18:
I have but two iSpindles and flashed the config. However it does not seem to create a Wifi AP that I can connect to. I get this  if I connect via USB and monitor the output.

Double Reset detected
...........
ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

Does anyone have any ideas what it could be.

camjose, 2020-08-20:
Hi, Same for me, see #413

chesterweirdo, 2020-08-20:
Yep that does look the same. I will have to also checking I get a temp setting.

ErikdBr, 2020-10-19:
> I have but two iSpindles and flashed the config. However it does not seem to create a Wifi AP that I can connect to. I get this if I connect via USB and monitor the output.
............
> started Portal
> *WM: SET AP
> *WM:
> *WM: Configuring access point...
> *WM: iSpindel
> *WM: AP IP address:
> *WM: 192.168.4.1
> *WM: HTTP server started
> 
> Does anyone have any ideas what it could be.

It has in fact created a wifi AP, you need to connect the wifi of a mobile device(phone, laptop, tablet) to the AP with the name 'iSpindel' and after you are connected go to the ip address 192.168.4.1 in a browser with the same mobile device.

nzpcon, 2020-10-24:
I'm having the exact same problem with some further code after a couple of minutes from WM HTTP server started.
Anyone have a fix?
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: After waiting...
*WM: 6.80
*WM: seconds
*WM: Timed out connection result:
*WM: WL_NO_SSID_AVAIL

Exception (3):
epc1=0x40100a25 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40005420 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff1190 end: 3fff1730 offset: 0190
3fff1320:  00000000 4026526d 3fff1b44 402164ca
3fff1330:  00000000 00000000 3fff1360 3fff1590
3fff1340:  3ffe8528 3ffe8528 00000020 40100ce3
3fff1350:  3ffe8528 00000006 3fff1680 402157c9
3fff1360:  3ffe8528 3fff1680 3fff1390 40211dfb
3fff1370:  3ffe8528 3ffe8528 00000000 40205f30
3fff1380:  00000000 feefeffe feefeffe feefeffe
3fff1390:  feefef00 feefeffe 80efeffe feefef00
3fff13a0:  feefeffe 80efeffe 00000000 3fff1b7c
3fff13b0:  3ffe9566 00000000 000493e0 00000000
3fff13c0:  00000f03 3ffe94e6 00000000 40105425
3fff13d0:  3fff13cc 402249a0 00000000 402249a0
3fff13e0:  00000000 402249a0 00000000 402249a0
3fff13f0:  00000000 402249a0 00000000 402249a0
3fff1400:  00000000 00000015 ffffffff 00000101
3fff1410:  3ffe9ddc 00000000 01000035 00000000
3fff1420:  40201df4 3fff1638 3fff1620 3fff1518
3fff1430:  3fff14d0 3fff1500 3fff14e8 3fff14b8
3fff1440:  3fff1668 3fff1650 3fff1608 3fff15f0
3fff1450:  3fff15d8 3fff15c0 3fff15a8 3fff1590
3fff1460:  3fff1578 3fff1560 3fff1548 3fff1530
3fff1470:  3fff14a0 3fff1488 0000001f 40100190
3fff1480:  40219ba0 2c9f0300 3ffe8d73 3ffe8d79
3fff1490:  3fff2dc4 000000c8 00000001 00000000
3fff14a0:  00000000 00000000 00000000 00000000
3fff14b0:  00000001 3ffe8d15 00000000 00000000
3fff14c0:  00000000 00000000 00000001 3ffe8ceb
3fff14d0:  00000000 00000000 00000000 00000000
3fff14e0:  00000001 3ffe8cbf 3ffe8cb5 3ffe8cb5
3fff14f0:  3fff2d24 00000005 00000000 3ffe8e99
3fff1500:  00000000 00000000 00000000 00000000
3fff1510:  00000001 40260ece 3ffe8c95 3ffe8c9b
3fff1520:  3fff1e9c 00000007 00000001 3ffe8e80
3fff1530:  3ffe8c8c 3ffe8c81 3fff2cf4 00000028
3fff1540:  00000001 3ffe9ddc 3ffe8c7d 3ffe8c72
3fff1550:  3fff2cc4 00000028 00000001 3ffe9ddc
3fff1560:  3ffe8c69 3ffe8b54 3fff2c94 00000028
3fff1570:  00000001 3ffe9ddc 3ffe8c60 3ffe8b4b
3fff1580:  3fff2c64 00000028 00000001 3ffe9ddc
3fff1590:  3ffe8c5d 3ffe8c54 3fff2c34 00000028
3fff15a0:  00000001 3ffe9ddc 3ffe8c45 3ffe8c49
3fff15b0:  3fff2b2c 000000fd 00000001 3ffe9ddc
3fff15c0:  3ffe8c2f 3ffe8c37 3fff1e6c 00000028
3fff15d0:  00000001 3ffe8e80 3ffe8c1e 3ffe8c23
3fff15e0:  3fff1e3c 00000028 00000001 3ffe8e80
3fff15f0:  3ffe8c08 3ffe8c0f 3fff2a24 000000fd
3fff1600:  00000001 3ffe9ddc 3ffe8c02 3ffe8b34
3fff1610:  3fff297c 000000a0 00000001 3ffe9ddc
3fff1620:  3ffe8be8 3ffe8bee 3fff1e2c 00000006
3fff1630:  00000001 3ffe8e80 3ffe8b4f 3ffe8bda
3fff1640:  3fff2924 00000050 00000001 3ffe9ddc
3fff1650:  3ffe8bd3 3ffe8bd3 3fff1e0c 00000014
3fff1660:  00000000 3ffe8e99 00000000 00000000
3fff1670:  00000000 00000000 00000001 40260f5f
3fff1680:  2e313900 00003038 80003634 3fff2d00
3fff1690:  0021002f 80000000 00000000 3fff00d4
3fff16a0:  3ffeff84 00000000 3ffefdd8 3ffefd88
3fff16b0:  3ffeff84 3ffefde7 3ffeffa4 40207056
3fff16c0:  00000000 00000000 00000000 00000000
3fff16d0:  feefeffe feefeffe 00000000 feefeffe
3fff16e0:  00000000 feefeffe feefeffe feefeffe
3fff16f0:  feefeffe feefeffe feefeffe feefeffe
3fff1700:  feefeffe feefeffe feefeffe 3fff00d4
3fff1710:  3fffdad0 00000000 3fff0094 40216db0
3fff1720:  feefeffe feefeffe 3ffe865c 40100fe5
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

   FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
ERROR: failed to load json config

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 2C 53 7 D6 1 3C 99
  Chip = DS18B20
  Data = 1 7E 1 55 0 7F A5 A5 0 14  CRC=14
  Temperature = 23.88 Celsius,
offsets not available
Boot-Mode: Exception

Double Reset detected

going to Config Mode
reboot RFCAL
;ld▒▒<▒l▒|
▒$▒c|▒▒▒{▒c▒c▒▒og▒log▒▒▒cx▒▒lrdrd

thegreatgunbantoad, 2020-10-25:
From first switch on leave the device alone for 10 seconds (This is for a freshly blanked and reprogrammed Wemos) to allow it to create it's config file.  After that the blue led should flash every secondish.  If it's flashing like that there should be an AP, it can take a while (up to a minute or so) for some devices to find the network btw.

Config files can get corrupted, in that case you need to erase and reflash the Wemos (The erase is the important bit there).

nzpcon, 2020-10-26:
> From first switch on leave the device alone for 10 seconds (This is for a freshly blanked and reprogrammed Wemos) to allow it to create it's config file. After that the blue led should flash every [...]

Thanks for the rsponse.
I've tried erasing & reflashing, but still no AP visible on wifi.
I have to do the double reset after power on to get the led to start flashing .

closed 2021-01-01 by stale[bot]