universam1/iSpindel #419: Cant flash wemos d1 mini.

helcik, 2020-08-23:
## I'm submitting a ...
  - [ ] bug report
  - [ ] feature request
  - [x ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [x ] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x ] Checked the logs for errors in the terminal console ?
- [ x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

I am having trouble flashing firmware on one of the wemos. I have done already couple of them but this one seems to not accept any way. I have tried to connect via putty but after pressing reset button, putty doesnt show anything. 

I have tried to flash it with Nodemcu flasher but no luck either with  the option to reset it totaly. When i conect the wemos to usb the light is constantly blue. I tried without battery and with battery. No joy.

Any ideas or is it broken wemos ? 
### Console Logs

Cant get them :/

thegreatgunbantoad, 2020-08-23:
Probably a poor quality clone board.  Some of them are so bad components are the wrong way around or even missing.

helcik, 2020-08-26:
Problem is that out of 10 wemos i have issues with 4 of them. So i was thinking if i am doing something wrong

thegreatgunbantoad, 2020-08-26:
4 out of 10 is bad, must be clone ESP board.  But if you can get the programming in general you probably have 4 naff boards.  Contact your seller.  I've had a number of the clone boards but never that bad.  Do be careful some of the clones have an issue that means they don't wake up after the first read.  If you're building to sell on it might look fine but not send to Brewfather consistently.

closed 2020-11-01 by stale[bot]