universam1/iSpindel #420: 8 day battery life, suspected not entering deep sleep, otherwise operating as expected.

bonley, 2020-08-26:
## I'm submitting a ...
  - [ ] support request

## Report

Battery lasting 8 days regardless of update interval and with a verified 3350mAh Panasonic NCR18650B. Logging current consumption, peaks at 99mA, averages 80mA during active periods. 11mA between updates. Long term average ~12mA, whic correlates strongly to the 8 day battery life. Plugged in via USB draws an average of 17mA and a baseline of 17mA between uploads, 85mA-90mA during uploads.

My current guess is that it doesn't enter deep sleep mode and just enters modem sleep mode instead, but I'm using Firmware 6.3.1, so if that were the case, I would expect the same to happen to others, unless it's a hardware fault.

Also worryingly draws 0.064mA while the switch is in the off position, which doesn't seem right. Probing the switch reveals 1-2Ohm closed and ~2.9MOhm open. I would expect more like 65kOhm for that current.

### Console Logs

FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"0.000449952*tilt^3-0.062310147*tilt^2+3.116627549*tilt -46.25709004","SSID":"","PSK":"","Offset":[3730,-436,1346,193,-176,53]}
applying offsets: 3730:-436:1346
confirming offsets: 3730:-436:1346
Boot-Mode: External System
power-cycle or reset detected, config mode

going to Config Mode
reboot RFCAL
FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"0.000449952*tilt^3-0.062310147*tilt^2+3.116627549*tilt -46.25709004","SSID":"","PSK":"","Offset":[3730,-436,1346,193,-176,53]}
applying offsets: 3730:-436:1346
confirming offsets: 3730:-436:1346
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:40 min:65.75 max:65.90 time:694
x: -13538 y: 8830 z: 16
Tilt: 65.80
Tacc: 26.57
Volt: 4.69
Temp: 25.06
Gravity: 17.22
After waiting 0ms, result 0
Failed to connect -> trying to restore connection...
Rescued Wifi credentials
   -> waited for 7384ms, result 3
   -> Connection restored!
Final-sleep: 900s; RT: 8600

FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"0.000449952*tilt^3-0.062310147*tilt^2+3.116627549*tilt -46.25709004","SSID":"","PSK":"","Offset":[3730,-436,1346,193,-176,53]}
applying offsets: 3730:-436:1346
confirming offsets: 3730:-436:1346
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:71.09 max:71.17 time:746
x: -14912 y: 6184 z: -168
Tilt: 71.12
Tacc: 25.72
Volt: 4.69
Temp: 24.69
Gravity: 22.09
After waiting 3001ms, result 3
IP: 192.168.1.21

calling Ubidots
{"tilt":71.12096,"temperature":24.6875,"battery":4.692388,"gravity":22.0911,"interval":900,"RSSI":-75}
Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel?token= HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 102

HTTP/1.1 200 OK
Server: nginx
Date: Wed, 26 Aug 2020 21:21:35 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Allow: GET, POST, HEAD, OPTIONS
Vary: Origin, Cookie

d1
{"battery": [{"status_code": 201}], "gravity": [{"status_code": 201}], "interval": [{"status_code": 201}], "rssi": [{"status_code": 201}], "temperature": [{"status_code": 201}], "tilt": [{"status_code": 201}]}
0

Final-sleep: 900s; RT: 5933

FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"0.000449952*tilt^3-0.062310147*tilt^2+3.116627549*tilt -46.25709004","SSID":"","PSK":"","Offset":[3730,-436,1346,193,-176,53]}
applying offsets: 3730:-436:1346
confirming offsets: 3730:-436:1346
Boot-Mode: Deep-Sleep Wake

woken from deepsleep, normal mode
Samples:42 min:71.32 max:71.43 time:746
x: -14956 y: 6070 z: -82
Tilt: 71.38
Tacc: 25.64
Volt: 4.69
Temp: 24.69
Gravity: 22.38
After waiting 2601ms, result 3
IP: 192.168.1.21

calling Ubidots
{"tilt":71.38267,"temperature":24.6875,"battery":4.692388,"gravity":22.3764,"interval":900,"RSSI":-73}
Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel?token= HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 102

HTTP/1.1 200 OK
Server: nginx
Date: Wed, 26 Aug 2020 21:35:53 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Allow: GET, POST, HEAD, OPTIONS
Vary: Origin, Cookie

d1
{"battery": [{"status_code": 201}], "gravity": [{"status_code": 201}], "interval": [{"status_code": 201}], "rssi": [{"status_code": 201}], "temperature": [{"status_code": 201}], "tilt": [{"status_code": 201}]}
0

Final-sleep: 900s; RT: 5289

## Context
Just built from The Jeffrey V2.1 kit from oh-hiuk. 

The worry is that this is all due to complications during building: I didn't flash the Wemos D1 prior to building but did remove the diode. Then took the battery out to flash it, which I now understand why this wouldn't work. So I desoldered it, resoldered the diode, flashed it, then put it back together with the diode still in place. Just need to make sure I remove the battery before plugging in the Wemos every time in future. The desoldering and resoldering was not smooth and a trace peeled off, but was still in tact. I carefully soldered it back together and it seems to work fine. Wondering if there's a hardware error that would present these symptoms? I've checked continuity between the GPIO16 pin on the Wemos and the BAT43 diode, and the RST pin on the Wemos to the other side of the BAT43 diode.

## Your Environment
- Firmware Version used: 6.3.1
- Service backend you have configured the iSpindel to talk to: Ubidots

universam1, 2020-08-27:
Did you remove the LED from the accelerometer?

bonley, 2020-08-27:
Yes, the LED is removed from the accelerometer.

bonley, 2020-08-31:
Turns out it was the Wemos D1 Mini. No idea exactly what was wrong with it, but swapping it out reduced my deep sleep power consumption to <1.5mA. My TP4056 Also started dropping 3.3V of the available 3.9V leaving only 0.6V for the Wemos, so it wouldn't even turn on. I replaced that too. It should last at least 10 times longer now. I predict about 80 days.

closed 2020-10-30 by bonley