universam1/iSpindel #423: iSpindel lying flat in pure water

jameshope87, 2020-09-21:
I have just finished constructing my iSpindel using the 3D printed sled, when I drop it into water it less almost flat, about 85 degrees.

This is with 30g of coins in the bottom. I cannot fit any more. What have I done wrong?

ErikdBr, 2020-09-21:
That is impossible to answer without detailed pictures.

jameshope87, 2020-09-22:
Sorry, I have updated my comment with some photos

ErikdBr, 2020-09-22:
I am not familiair with this design, but shouldn't it float with the battery down, now the coins are counterweight for the battery. If you shift the coins to the side of the battery it adds to the weight of the battery.

ErikdBr, 2020-09-22:
I have an earlier design of https://www.3d-mechatronics.de/ (not the one which is on there now) with a customized designed sled(which you can find in the universam ispindel project somewhere) and in the picture you can see that my battery is pointed downwards:
[image]

jameshope87, 2020-09-22:
After adjusting I managed to get it to flat battery side down, and then with as many coins in as I could, 6 5ps, 3 1ps and 2 2ps, I finally managed to get it approaching 20 off vertical. But the info screen read 60, this seemed to be the lowest angle it would read, 89.98 when horizontal and then down to about 60 odd when vertical.

That was before the tube filled with smoke as the battery wires vaporised. I'm not having a good day it seems...

thegreatgunbantoad, 2020-09-22:
How heavy is the battery.  Lot of fakes out there, if it's under 40g throw it a long long way away.

jameshope87, 2020-09-22:
It is only 33.88g - I will place an order for a Panasonic or similar and see if that fixes it.

ErikdBr, 2020-09-22:
I use the Samsung INR18650-35E but again I have a different design.
[image]

jameshope87, 2020-09-23:
Well I might try printing a new sled, this one seems annoying!

How did you get those solder tabs on your cell? It's seems like soldering directly to the cell would be difficult

ErikdBr, 2020-09-23:
> ...That was before the tube filled with smoke as the battery wires vaporised. I'm not having a good day it seems...

Well good that it wasn't in a brew allready!

You can order these batteries with or without the tabs. But these are unprotected batteries so you'll have to be 100% sure that your tp4056 module is the wright one and working properly and make no mistakes soldering the parts together otherwise you can end up with an exploding iSpindel or another fire.

jameshope87, 2020-09-23:
Thank you very much, it seems UK suppliers are few and far between, but hopefully will get one soon.

I seem to have managed to trace my exploding battery wires to a faulty Wemos. No short when the Wemos is not connected, but there is a 75kO resistance between the battery terminals with the Wemos mounted. DMM confirms a short on the Wemos board so have ordered some replacements as I can't find out where on the board it is!

ErikdBr, 2020-09-23:
You really don't need 3500 mAh(that is just my fear of running out of power lol) , 3000 mAh will do just fine: [https://18650.uk/product/samsung-30q-button-top-18650-battery/](url) I did a little search for UK suppliers but couldn't find any with soldering tabs. But the one in the link is with a button top so you can put it in a 18650 battery holder, still not protected though.

closed 2020-11-29 by stale[bot]