universam1/iSpindel #424: Issues connecting iSpindel to internet

projects-in-progress, 2020-09-28:
I'm submitting a support request after following the FAQ and searching issues log for similar issues. I'm having issues connecting the iSpindel to my network successfully and repeatably.

### Console Logs
This is a snipet of a successful connection: 
18:23:01.493 -> Tilt: 89.23
18:23:01.493 -> Tacc: 27.68
18:23:01.493 -> Volt: 4.61
18:23:01.532 -> Temp: 33.94
18:23:01.532 -> Gravity: 33.18
18:23:01.532 -> IP: 10.0.0.141
18:23:01.532 -> 
18:23:01.532 -> calling HTTP

typically I get 1 successful connection, and then every connection after fails with a log similar to this
18:36:39.997 -> Tilt: 88.88
18:36:39.997 -> Tacc: 26.44
18:36:39.997 -> Volt: 4.61
18:36:40.032 -> Temp: 33.69
18:36:40.032 -> Gravity: 33.00
18:36:40.032 -> After waiting 10204ms, result 6
18:36:50.244 -> Failed to connect -> trying to restore connection...
18:36:50.382 -> Rescued Wifi credentials
18:36:50.382 ->    -> waited for 10994ms, result 6
18:37:01.366 ->    -> Failed to restore connection...
18:37:01.366 -> Final-sleep: 910s; RT: 22433

Occasionally this will be displayed in the log after an unsuccessful connection:
18:52:13.243 -> Tilt: 86.86
18:52:13.243 -> Tacc: 26.24
18:52:13.243 -> Volt: 4.61
18:52:13.277 -> Temp: 33.63
18:52:13.277 -> Gravity: 31.99
18:52:13.277 -> After waiting 10204ms, result 6
18:52:23.482 -> Failed to connect -> trying to restore connection...
18:52:23.621 -> Rescued Wifi credentials
18:52:23.621 ->    -> waited for 9993ms, result 3
18:52:33.600 ->    -> Connection restored!
18:52:33.600 -> Final-sleep: 910s; RT: 21437

Interestingly. when the iSpindel is put into config mode it seems to function perfectly, and it’s the first connection after exiting the config page that is successful.

## Your Environment

- Firmware Version used: 6.3.1
- Browser Name and version: Google Chrome
- Service backend you have configured the iSpindel to talk to: same issue with both Grainfather connect and Brewfather software

I've tried on a few different wifi networks and backends, with no difference in the results. I've left the iSpindel for a period of about 24 hours and had zero succesfull after the first one connections.

I'm sure I must have missed something simple here for it to consistently work once and then not again. Does anyone have suggestions for what I should be looking for?

Pertti75, 2020-09-28:
I sounds that have this same issue with 6.3.1, Chrome and Brewfather, but I don't have logs currently.
It only sends data to Brewfather after first connection after config or if I press reset button. First I used 6.2.0 firmware as it had this issue then I tried to solve it with updating to 6.3.1, but without success.
Seems that I am not even able to change connection anymore to different wlan as it seems to stuck in same connection all the time..

thegreatgunbantoad, 2020-09-28:
If it's only sending one reading it's probably failing to wake from sleep.  Check the diode is the right way around.  If it is it's probably a duff Wemos/ESP board.

projects-in-progress, 2020-09-29:
I checked the diode, no issues there. I also enabled MAC address filtering in my router for the ispindel and managed to get almost 24 hours of successful connections but it stopped transmitting a few hours ago. I've plugged it in to charge now, thinking the battery might be a bit flat, if it doesn't start to transmit again with acharged battery I"ll reconnect it to the Audrino IDE and look at the logs with the serial monitor.

projects-in-progress, 2020-09-29:
Logs from the latest failed connection:
16:51:25.541 -> Tilt: 84.99
16:51:25.541 -> Tacc: 27.59
16:51:25.541 -> Volt: 4.61
16:51:25.576 -> Temp: 35.06
16:51:25.576 -> Gravity: 31.04
16:51:25.576 -> After waiting 10203ms, result 6
16:51:35.790 -> Failed to connect -> trying to restore connection...
16:51:35.933 -> Rescued Wifi credentials
16:51:35.933 ->    -> waited for 10993ms, result 6
16:51:46.906 ->    -> Failed to restore connection...
16:51:46.906 -> Final-sleep: 1200s; RT: 22429

Pertti75, 2020-09-29:
> If it's only sending one reading it's probably failing to wake from sleep. Check the diode is the right way around. If it is it's probably a duff Wemos/ESP board.

Diode is right way, but seems that I have an issue with diode soldering as it was not attached to D0. After re soldering I am able to see that it is waking up and sending information to Brewfather. Thank you very much for your help!
But I am not still able to change wlan connection to other one as it sees to stick in same connection all the time as it won't save new wlan connection if it is changed from config.

projects-in-progress, 2020-09-30:
Still having issues with this, it transmitted successfully for about 18 hours and then stopped 7 hours ago. Now hen I look at the logs I get a ismilar result to before:

18:26:00.110 -> woken from deepsleep, normal mode
18:26:00.863 -> Samples:42 min:74.80 max:74.96 time:749
18:26:00.863 -> x: 14308 y: 4384 z: -164
18:26:00.863 -> Tilt: 74.88
18:26:00.863 -> Tacc: 23.73
18:26:00.863 -> Volt: 4.19
18:26:00.897 -> Temp: 29.44
18:26:00.897 -> Gravity: 25.92
18:26:00.897 -> After waiting 10204ms, result 6
18:26:11.077 -> Failed to connect -> trying to restore connection...
18:26:11.214 -> Rescued Wifi credentials
18:26:11.214 ->    -> waited for 10995ms, result 6
18:26:22.217 ->    -> Failed to restore connection...
18:26:22.217 -> Final-sleep: 1200s; RT: 22425

But when I look at my routers page I can see it registering as a wifi device for about 4 seconds twice each time. I can't see anything in the logs from my router to see whats happening either.

Unsuccessful connection
Sep 30 18:44:14   wlceventd: WLCEVENTD wlceventd_proc_event(500): eth5: Auth 52:6G:AD:9C:ZZ:FE,   status: Successful (0)
--
Sep 30 18:44:14 wlceventd: WLCEVENTD   wlceventd_proc_event(529): eth5: Assoc 52:6G:AD:9C:ZZ:FE, status: Successful   (0)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(481): eth5: Disassoc 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Disassociated because sending station is leaving (or has left) BSS   (8)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 18:44:21 wlceventd: WLCEVENTD   wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE, status: 0,   reason: Class 2 frame received from nonauthenticated station (6)

Successful connection 
Sep 30 02:58:48 wlceventd:   WLCEVENTD wlceventd_proc_event(500): eth5: Auth 52:6G:AD:9C:ZZ:FE, status:   Successful (0)
Sep 30 02:58:48 wlceventd:   WLCEVENTD wlceventd_proc_event(529): eth5: Assoc 52:6G:AD:9C:ZZ:FE, status:   Successful (0)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(481): eth5: Disassoc 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Disassociated because sending station is leaving (or has   left) BSS (8)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:58:49 wlceventd:   WLCEVENTD wlceventd_proc_event(466): eth5: Deauth_ind 52:6G:AD:9C:ZZ:FE,   status: 0, reason: Class 2 frame received from nonauthenticated station (6)
Sep 30 02:59:02 ahs:   [read_json]Update ahs JSON file.
Sep 30 02:59:32 ahs: [read_json]Update   ahs JSON file

projects-in-progress, 2020-10-01:
I'm starting to think it's a dodgy wemos board, just based on the router telling me the connection was successful.

thegreatgunbantoad, 2020-10-03:
The JSON file can get corrupted.  Try Erasing the ESP/Wemos board and re-flashing.  You need to do the erase too, that's the important bit, not just a reflash.  This tool will do it: https://github.com/thorrak/brewflasher/releases/tag/v1.0.1

themoch, 2020-12-02:
@projects-in-progress I have had 2 batches of boards be either knock offs and not work right, or just bad.  I'm hesitant to pay higher prices for boards not knowing if they are going to be quality or not.  The boards I purchased were not inexpensive either.

I think it's just bad batches of hardware

ErikdBr, 2020-12-02:
> I have a PCB with sockets fitted so I can test easily

A bit off topic but do you want to show me how you set this up(with pics) so I can test my boards too before assembling.

Allthough I buy the same everytime on aliexpress (https://nl.aliexpress.com/item/32529101036.html) where I didn't have one falty yet out of 15 or so, these are fairly expensive compared to the clones on aliexpress. And I want to try these cheaper clones which cost about half.

themoch, 2020-12-02:
@thegreatgunbantoad i too would like to see this setup. 

Do you have the board rigged with female sockets and then just solder the large pins to the wemos d1mini?

thegreatgunbantoad, 2020-12-02:
Used the sockets that came with I think the clone Wemos kits

[image]
[image]

thegreatgunbantoad, 2020-12-02:
That's my special Wemos board with a reed switch in the reset line so you can in theory seal the unit up forever.  (One caveat the battery will just run out needs another tweak to stop that)

closed 2021-02-08 by stale[bot]