universam1/iSpindel #425: iSpindel reporting extremely low voltage.

themoch, 2020-10-07:
## I'm submitting a ...
  - [ ] bug report
  - [ ] feature request
  - [x] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [x] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

iSpindel shows output of voltage at `0.036496` while multimeter reads `3.89` and this causes unit to go into safe mode, broadcasting every hour.

## Context
I have tried multiple working batteries from other units to no success.

As stated in the [FAQ ](https://github.com/universam1/iSpindel/blob/8150afa9bc7ee4ea9e1e13052880b853e71f9fcd/docs/FAQ-en.md#questions-and-answers-faq) I need to update my voltage factor in order to make the battery read correctly.  

When following the formula I need to set the value to approximately `1.24`.  Because of the following lines of code it will not save those values:

[iSpindel.cpp](https://github.com/universam1/iSpindel/blob/2054d44e5a3194318be38a646aa83ca835be9eb0/pio/src/iSpindel.cpp#L422-L423)

```
 if (my_vfact < ADCDIVISOR * 0.8 || my_vfact > ADCDIVISOR * 1.25)
    my_vfact = ADCDIVISOR;
```

I have used the multimeter to check that I have the correct resistors in place AND they are reading at the proper resistance of 4.7K and 220K.  The diode is oriented in the proper direction and is reading ~`240`

## Your Environment

- Firmware Version used: `6.3.1`
- Service backend you have configured the iSpindel to talk to: `personal http endpoint logging` and Arduino serial monitor

thegreatgunbantoad, 2020-10-07:
If you're getting a voltage like that there is likely something wrong with the build.  I would suggest checking your connections of the 220k right to the pin of the metal can on the ESP/Wemos board and to the pin of the Lipo Charge board.  You're basically seeing 0 volts so suggests you have an iffy connection.  Possibly a duff ESp/Wemos.

themoch, 2020-10-07:
> If you're getting a voltage like that there is likely something wrong with the build. I would suggest checking your connections of the 220k right to the pin of the metal can on the ESP/Wemos board [...]

I’m getting the correct readings at the polls on the d1mini. I’m suspecting that the batch of wemos I got are all bad. I have a few others with weird issues.

thegreatgunbantoad, 2020-10-08:
A whole duff batch is pretty unlucky.  I find about 1 in 10 of the Wemos clone boards on eBay are duff.  For a while the Accelerometer boards were about 1 in 5 was duff, but that blight seems to have passed.

themoch, 2020-10-08:
@thegreatgunbantoad It’s about 3/5 at this point. I have 2 others that act normal until you go to connect via wifi. Then nothing can connect to them. 

One other that connects to the AP only about 20% of the time. The rest of the time it times out after 10 sec of trying. Sometimes that one says it's successfully connected, but then doesn't send data.

¯\_(ツ)_/¯

What's your source, maybe I'll try ordering from there?

thegreatgunbantoad, 2020-10-25:
@themoch I just grab stuff off eBay and test everything.  I have a PCB with sockets on it so I can use that as a testbed for the various parts and just plug then in and out as needed.

universam1, 2020-12-20:
Not firmware related

closed 2020-12-20 by universam1