universam1/iSpindel #426: Angle to steep in plain water? (not an issue, just a question)

ErikdBr, 2020-10-16:
Because I have a 3 week vacation and I am at home instead of in Thailand enjoying the sceneries and eating great food and drinking Thai beer, hug elephants, get amazing massages etc. because all flights are cancelled and the alternatively booked week in a bungalow in Roompot Cochem Germany is canceled too because of Covid-19 I finally got time to build a brand new iSpindel because I have a ton of parts laying around.

But now the thing, when the build was ready I poured 20 ml of epoxy in the tube before putting in the sled to secure the sled in the tube so it can not move anymore and you only have to calibrate once. Because I already suspected 20 ml is to much I left out 2 washers that I do have in a previous build. Afterwards I realised it had to be only 10 ml of epoxy.

So now when I put it in plain water it has an angle of 9,3 degrees.

Is it still good to go or do I have to saw that thing out of there?
[image]

RonnyRusten, 2020-10-16:
I don't know if the angle will cause problems, but instead of sawing it out of there, what about just sawing of the tip?

ErikdBr, 2020-10-17:
Thanks, that is actually a good idea... but the inside of the thread where the bolt is on is hollow, so then I have to fill that up with epoxy and I have to get the bottom smoother then a baby's buttocks :-) to avoid contamination so I rather would here from someone, maybe @universam1 himself, if the angle of 9,3 is a problem or not.

thegreatgunbantoad, 2020-10-25:
Down that end the polynomial curve is a little steeper, so for each degree of tilt there is a larger gravity swing.  Essentially it will be a little oversensitive around the water reading, a slight difference in tilt will translate to a larger difference in gravity.  You'll definitely need to use a 3rd order polynomial at 9.3 degrees.  (Try this: https://github.com/universam1/iSpindel/blob/master/docs/iSpindelPolyEstimator.xlsx)

I suspect @universam1 chose 25ish so that your typical brewing gravities all sit in the straighter section of the curve.

closed 2020-12-31 by stale[bot]