universam1/iSpindel #429: Cannot calibrate 0 offset, just hangs.  No tilt value in Summary screen.

tomohulk, 2020-10-28:
## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [x] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [ ] Checked the logs for errors in the terminal console ?
- [ ] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

Using firmware 6.3.1, i cannot calibrate the 0 offiset, the blue light stays on and it seems froze.  i ran through multiple times using the suggestions from the FAQ and from the link at the bottom, following others suggestions.  It seems like it just is not flashing properly.

### Console Logs
Here are some excerpts from the console log, but seems like the `Failed to mount FS!` and the `Acc Test Connection ERROR!` are bad.  I have two iSpindels that are reporting the same exact things.  I also am not seeing any tilt info in the summary screen, but when i connect it to ubidots i do get that info.  One other thing i wiould like to add is if i go to an older firmware 6.1.3, i can see the tilt value in the summary screen, but i still get the freezing when doing the 0 offset calibration.

Initial boot after flash:
```
          FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 FF E2 CC 7A 18 1 7B
  Chip = DS18B20
  Data = 1 63 1 4B 46 7F FF C 10 D1  CRC=D1
  Temperature = 22.19 Celsius,
Acc Test Connection ERROR!
offsets not available
Boot-Mode: Deep-Sleep Wake
```

Running 0 calibration offset:

## Steps to Reproduce (for bugs)

1.  Flash iSpindle with NodeMCU manager using detailed steps from FAQ when the device is not working correctly after flashing OTA.
2. Attempt to calibrate 0 offset.
3. Steps never finish

## Context

Just trying to do an initial setup and calibration.

## Your Environment

- Firmware Version used: 6.3.1
- Browser Name and version: Safari on iPadOS 14 
- Service backend you have configured the iSpindel to talk to: none as of yet, still just trying to calibrate.

## Features

**Please note by far the quickest way to get a new feature is to file a Pull Request.**

ErikdBr, 2020-10-28:
D1 mini has 4mb memory. Maybe that's your problem?

nkaramolegos, 2020-11-22:
So do you find the problem? I have the some.

tomohulk, 2020-11-23:
@nkaramolegos I haven't messed with them in a few weeks, my guess is the gyro boards though.

ErikdBr, 2020-11-23:
> @nkaramolegos I haven't messed with them in a few weeks, my guess is the gyro boards though.

As I stated before, the d1 mini has 4MB memory, not 1MB. These are my settings when flashing without any hassle:
Baudrate: 115200
Flash size: **4MByte**
Flash speed: 40MHz
SPI Mode: **DIO**

tomohulk, 2020-11-23:
well, I feel like an idiot now @ErikdBr lol.  for some reason I read your message the other way and was thinking the flash size was 4MB and the board I had only had 1MB.  Ill try this when I get some time and report back.  Thanks.

tomohulk, 2020-11-23:
Eh, same issue.  I even did the reset first. 

```
		1st column		2nd column
1st Line:	INTERNAL://NODEMCU	0x00000
2nd Line:	INTERNAL://BLANK	0x01000
```
Still shows 0.00 for tilt angle and the calibration just hangs and hangs.

nkaramolegos, 2020-11-23:
Same here my board has 4MB and the problem remains

thegreatgunbantoad, 2020-11-24:
You may have ended up with a corrupt config file (json) and might need to do an erase of the Wemos.  I have had issues before that were only cleared by an erase not just a reflash.  You can do that either using VS Code/Platform IO and lording the project or a bit more simply using this tool: https://github.com/thorrak/brewflasher/releases/tag/v1.0.1 (It will erase and reload the latest firmware)

nkaramolegos, 2020-11-24:
I am on linux. I have tried python3 esptool.py --port /dev/ttyUSB0 erase_flash but didn't help. I will check your proposal but I have to find something similar for linux

nkaramolegos, 2020-11-24:
For now I made a tricks and seems that it works. I press the calibrate button I wait for 10 sec and then I disconnect from the wifi. I wait for 1 min and then I connect to wifi again and visit the main page of ispindel. In info tab I can see that the calibration was sucess (near 90 degrees). I am using the v6.2 of firmware

tomohulk, 2020-11-24:
thanks @thegreatgunbantoad the erase defiantly worked as my wifi settings are gone, but same issue, the calibration just never finishes, the blue light never goes back to flashing.  ill try with 6.2 when I get some more time.  can't do it as easy with that tool because its not the latest release.

thegreatgunbantoad, 2020-11-24:
Worth trying out the new 6.5.0 pre release.  Does sound like your gyro might be bust though, had a few duff ones myself.

nkaramolegos, 2020-11-24:
I tried 6.5 and sometimes works but always. I was trying to made the 25 degrees calibration but I don't know what should i check. The til value or the gravity? I move the ispindel but never the tilts value reach 25 degrees

thegreatgunbantoad, 2020-11-25:
After you do the sensor cal with the unit flat (Horizontal) Gyro pointing up, you want to be at 25 deg tilt thereabouts in tap water.  Don't get obsessed with a perfect 25, +-5 deg is fine.  I've got units that are down at 19ish and work fine.  After you have it sitting right in water, grab the SG of your water (my tap water is not 1.000 dead if it's cold) and the tilt reading then try: https://github.com/universam1/iSpindel/blob/master/docs/iSpindelPolyEstimator.xlsx for your polynomial.

RonnyRusten, 2020-11-25:
> After you do the sensor cal with the unit flat (Horizontal) Gyro pointing up, you want to be at 25 deg tilt thereabouts in tap water. Don't get obsessed with a perfect 25, +-5 deg is fine. I've got [...]

This Poly estimator isn't working. The poly shows as #value, so does the 4 cells behind the chart...
[image]

thegreatgunbantoad, 2020-11-25:
What program did you use to open it?  It should work in Excel or Google Sheets.

RonnyRusten, 2020-11-25:
I opened it with Excel. When I open it, there is a poly, but as soon as I change one of the values in the blue cells, it goes to #Verdi (#VALUE in Norwegian). It might perhaps have something to do with decimal symbol, I can play around a bit...

thegreatgunbantoad, 2020-11-25:
Those rows above #VERDI! showing a value should mean the #VERDI cell reads okay.  That cell is just a formatting function.  Ahh that might be it!  Cell F7 reads =TEXT(F6,"0.0000E-0"), but it might need you to change the decimal identifier there.

thegreatgunbantoad, 2020-11-25:
Can you try in F7 putting in =IFERROR(TEXT(F6,"0.0000E-0"),TEXT(F6,"0,0000E-0"))
That should make it fairly region independent hopefully.

thegreatgunbantoad, 2020-11-26:
I'm not sure the iSpindel code will handle Euro style decimals in the poly though.  Interesting challenge.

RonnyRusten, 2020-11-26:
I will look at it. As you say, the iSpindel probably needs a . (dot) as decimal separator in the poly, so the poly with , will probably cause an error.

nkaramolegos, 2020-11-29:
> @nkaramolegos I haven't messed with them in a few weeks, my guess is the gyro boards though.

I changed the gyro board and works fine

delight-f, 2021-01-04:
I'm having exactly the same problem. Cherryspindel board. I replaced the Wemos D1 mini, replaced the charging module and replaced the MPU6050. Have tried a total of four different MPU6050 gyroscopes.

When trying to calibrate, the blue light on the Wemos D1 mini just stays lit. I have to reboot the device manually. Tilt and gravity readings just read 'NaN'.

I've read the WHOAMI register values for all four devices and they are on spec at 0x68. I've tested one of the modules with some arduino code and it is reading all the appropriate values for the gyroscope. I've also tested for continuity between the SDA/SCL pinouts from the MPU6050 to the D3/D4 pins on the Wemos.

Have also flashed 6.5.0 with no gain.

delight-f, 2021-01-04:
So I actually edited the code as above and the device works - I get a reading. It cannot complete the offset calibration, however. On the horizontal, it reads 87 degrees. I can move the device in all planes and it changes appropriately - vertically, it reads 176 degrees.

mross286, 2021-01-16:
All your MPU boards have corrupted register. That's it. Maybe chinese fake or defective.
The WHO_AM_I  address are the upper 6 bits of the MPU-6050’s 7-bit I2C address on the register 117 and the bit 0 and 7 are harcoded to zero and reserved. Because the i2c adress is 1101000 (0x68) with AD0 low, and 1101001 (0x69) with AD0 high, the WHO_AM_I have always to be 110100 (0x34), as stated at page 45 of the attached document.
There's usually calibration issues with this problematic boards.
Ciao!
[MPU-6000-Register-Map1.pdf](https://github.com/universam1/iSpindel/files/5825224/MPU-6000-Register-Map1.pdf)

pppedrillo, 2021-01-24:
> All your MPU boards have corrupted register. That's it. Maybe chinese fake or defective.
> The WHO_AM_I address are the upper 6 bits of the MPU-6050’s 7-bit I2C address on the register 117 and the bit [...]

Thats what mross286 said. Read my old comment and Sam's resolution for it. https://github.com/universam1/iSpindel/issues/302#issuecomment-545382102

ErikdBr, 2021-01-24:
Due to some dodgy boards that I bought I have taken up the habbit to test these boards before I solder them. You'll need an Arduino uno, a breadboard and some breadboard wires and a program from the Arduino's website called MPUTeapot. If they don't work I contact the seller for a refund, which I get up until now because I buy all my parts from aliexpress and they refund very easily if you have proof.
[image]
[image]
[image]

thegreatgunbantoad, 2021-01-25:
I test them too, I use a built up iSpindel PCB with some headers where the gyro and ESP/Wemos should be should be (Plenty of dodgy ESP boards out there too).
They can sometimes give a reading but still fail to show a correct reading over the full range, so do check that too.

ErikdBr, 2021-01-25:
> They can sometimes give a reading but still fail to show a correct reading over the full range, so do check that too.

Good tip, thanks!

thegreatgunbantoad, 2021-01-25:
Also worth mentioning the firmware 6.5.0 improves the accuracy of the reading if there is a bit of pitch/roll.
It's not the official release yet, but it works great.
https://github.com/universam1/iSpindel/releases/tag/6.5.0

closed 2021-04-02 by stale[bot]

nkaramolegos, 2021-06-09:
Does somebody has a verified link for MPU 6050 which are not dodgy?

ErikdBr, 2021-06-09:
> Does somebody has a verified link for MPU 6050 which are not dodgy?

The last ones I bought was in februari this year: https://nl.aliexpress.com/item/4001203301527.html
They worked fine but that is no guarantee for future orders.

I see that I paid €6,65 and free shipping. They now more then doubled in price plus € 3 for shipping.