universam1/iSpindel #430: Showing to much voltage?

savvas1802, 2020-11-06:
Hello today I have build my ispindel according to the instructions. I used the 1st variant( I made my own circuit board etc.) I used 220k resistors, 470ohm and a 4k7 ohm as instructed. Everything is working fine except for some reason the voltage of the battery is showing 5.34 volt but is 3.56 when measured with a multimeter. I can't adjust with the correction factor as it is more than 20%. What did I do wrong? Thanks.

ErikdBr, 2020-11-06:
I don't know if it has to do with your problem but the resistor 470Ω has been replaced by a 390Ω resistor or even better a BAT43 Schottkey diode, due to recovering from deep sleep issues.

savvas1802, 2020-11-06:
Thanks for the reply I will try that, 390 or 330ohm resistor?

ErikdBr, 2020-11-06:
330 I just read in the parts list.

savvas1802, 2020-11-06:
Nope I just replaced it, same issue. Ahh never mind it must be something maybe with the battery or coding? 
Because when I take a measurement from the memos it is the same as the battery voltage 3.55.

thegreatgunbantoad, 2020-11-16:
Have you tried measuring the voltage after the pot down?  There's a 100k/220k in the Wemos and the iSpindel adds 230k to the other 220k.  So you should be seeing Vbat * 100/(100+220+230).  Assuming your battery is around 4Vish (I know they claim to be 3V7 but that's not what I've seen them measure in at when fully charged) you should be getting about 0.72V.

closed 2021-01-22 by stale[bot]