universam1/iSpindel #432: Config file corruption

thegreatgunbantoad, 2020-11-12:
## I'm submitting a ...
  - [ ] feature request

## Report
Sometimes (especially on the clone ESP boards) the json config file gets corrupted, requiring a full erase and reflash.
Can an option be added somewhere to delete and regenerate a config file, similar to what I imagine happens upon switch on of a freshly erased programmed device.

I will look into a code update for this with a view to creating a pull request when I have working code, but if anyone has any info of how to enact that deletion, that would be handy.
I imagine I need to use SPIFFS.

thegreatgunbantoad, 2020-11-12:
Right so I've found that Maintenance > factory reset does do a formatSpiffs call.  That hasn't in my experience fixed the issue before.  I wonder that format is like a quick format on a HDD, rather than a complete reformat.

universam1, 2020-12-20:
Try using the latest firmware version and call factory reset under maintenance menu to format your spiffs. If that doesn’t help your device is bricked

closed 2020-12-20 by universam1

thegreatgunbantoad, 2020-12-20:
I have seen units fixed by a full erase rather than remained bricked.  However I do concede that this need not be dealt with in the software.