universam1/iSpindel #433: Unable to calibrate sensible values

jpclouduk, 2020-11-16:
I'm going to attach some logs once I get my IDE talking to wemos correctly on Debian. In the meantime, I've searched through all the bugs/faqs/etc and can't find anyone with this exact issue other than where the calibration is done when the accelo is NOT facing up.
- Completed the build
- Flashed without issue and hooked up to Prometheus pushgateway
- While setting up the dashboard I noticed the tilt angle was showing an angle which seems wrong.
I dropped back to setup mode and ran the calibration. With the accelerometer facing up (horizontal ispindel) I calibrated 89.99. Then I tested the tilt angle as I moved the ispindel to the vertical the tilt angle reached 45 when upright. as I continued through back to the opposite horizontal (accelo upside down) the tilt went past 0. Effectively it's reading 180deg as 90.
I initialized the wemos and reflashed and calibrated but still the same.

Any thoughts while I gather some useful info ?

jpclouduk, 2020-11-16:
Got the IDE working. here's the calibration from the logs;
`*WM: Request redirected to captive portal
*WM: offset
Testing device connections...
MPU6050 connection successful
>**......>......
at 600 Readings

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS    -1868,    -940,    3058,     122,      19,     -58

new offsets: -1868:-940:3058
confirming offsets: -1868:-940:3058
saving config...
{"Name":"iSpindel000","Token":"","Sleep":900,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":191.8,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"","PSK":"","Offset":[-1868,-940,3058,122,19,-58]}
saved successfully
*WM: Request redirected to captive portal`

jpclouduk, 2020-11-17:
Does anyone know an easy way to read the X,Y,Z values in arduino IDE? I think these values are probably causing the issue as they seem way out to some of the offset readings I have seen from others. I have another wemos/gy521 on order to compare.

thegreatgunbantoad, 2020-11-18:
You probably have a knackered accelerometer.  Had a few duds myself do very similar.

jpclouduk, 2020-11-20:
> You probably have a knackered accelerometer. Had a few duds myself do very similar.

Yeah, got the new wemos and accel today and rebuilt. All testing is fine but I noticed I'm now seeing the Deep Sleep issue where the board doesn't wake up after the first ping of metrics. Checked all the soldering and shorts etc but all good. I'll take the new accel off and put it on the old wemos. From what I have read there are some wemos boards that don't like the bat42 resistor but that's beyond me.  Very frustrating but that's DIY ;)

thegreatgunbantoad, 2020-11-23:
Make sure that BAT42 diode is the right way round, fitting it backwards will prevent wake from deep sleep.

ErikdBr, 2020-11-23:
> Make sure that BAT42 diode is the right way round, fitting it backwards will prevent wake from deep sleep.

Isn't it a BAT43 diode? I use it.

thegreatgunbantoad, 2020-11-24:
> > Make sure that BAT42 diode is the right way round, fitting it backwards will prevent wake from deep sleep.
> 
> Isn't it a BAT43 diode? As suggested here: [#59 (comment)](https://github.com/universam1/iSpindel/issues/59#issue-224394180)

It is indeed I just followed the same typo from the post before, doh!

jpclouduk, 2020-11-25:
> It is indeed I just followed the same typo from the post before, doh!

Many thanks. I switched to the old wemos with the new accelo and all working 100%.  I have quite a few components now so I'm going to breadboard everything and test before soldering. I'll retest the faulty wemos in the breadboard setup.

Feel free to close this. This does appear to be a hardware issue.

thegreatgunbantoad, 2020-11-25:
> Many thanks. I switched to the old wemos with the new accelo and all working 100%. I have quite a few components now so I'm going to breadboard everything and test before soldering. I'll retest the [...]

Usually wemos clone boards and the gyros are the ones that can be duff.  Only ever had 1 charger board go.

closed 2021-02-01 by stale[bot]