universam1/iSpindel #435: genericHTTP API post sends wifi SSID and password in plaintext...

andyjbm, 2020-11-28:
Hi there,
Correct me if I'm wrong but doesn't this and subsequent lines serialize the entire config and thus sends the WIFI ssid and password over the internet in plain text??? (As we are using http and not https!)

https://github.com/universam1/iSpindel/blob/1d13253ceb563c8393f638d03290c5a393b0ce4e/pio/lib/Sender/Sender.cpp#L224

I'm checking my internet security 101 manual but I'm pretty sure that's neither a good idea nor even necessary!

Regards,

Andy.

thegreatgunbantoad, 2020-11-28:
It only sends the stuff added in iSpindel.cpp I think (by sender.add).  It doesn't get the whole saved json doc in memory.  For HTTP it looks fine.  See iSpindel.cpp lines 620 to 658.

iSpindel.cpp line 571 for API_MQTT, 602 for API_INFLUXDB and 713 for API_BREWBLOX do seem to pass it through but you need to look into Sender.cpp functions for those to see what they do.

andyjbm, 2020-11-28:
Yes @thegreatgunbantoad  you're right.

I was confusing the doc variable in ispindel.ino with the private _doc variable in the sender class.

Only those key value pairs added using the sender class .add method are placed in _doc as you say.

I'm trying to post to an emoncms service and I'm ending up adding my own method because the generic http method doesn't handle emoncms node and apikey in the required way.

This can be closed - do you need me to do that?

Thanks.

Andy.

closed 2020-12-20 by universam1