universam1/iSpindel #436: Temperature Compensation, any one interested?

thegreatgunbantoad, 2020-12-01:
## I'm submitting a ...
  - [ ] feature request

## Features

I have temperature compensation working based on the calcs in the HTLM poly generator doc.  It doesn't do a lot, a couple of SG points if your at 30deg but calibrated at 20deg for example.  It needs some polish to add it to the main project, I've never bothered as I don't really care for it, just did it for the challenge.  Does anyone think it's worth adding, I might get my finger out and polish it up a bit.

quarky42, 2020-12-09:
I would really appreciate this being added.

Personally I find error to be more of a concern in the situation where sources of error are additive. 

Being able to toggle this compensation on or off, with it being off by default, might be ideal.   

Alternatively, having both the uncompensated and the temperature compensated value would let someone decide which value they want to use.   If I know there is CO2 present in the must, I'll go with the uncompensated value because the dissolved CO2 can throw the reading off quite a bit anyway.   

At the end of a batch when the yeast activity is tiny or non-existent, there should be an ever decreasing amount of CO2 present.   Also, the yeast aren't generating their own warmth so one of the temperature variables is simplified.   At that point I would like to see the compensated value.

So, yeah I see a use case for having both pieces of information available.

thegreatgunbantoad, 2020-12-11:
Sorry for the slow response, been working on getting a pressure senor to work.  Nearly there (code is still a bit unstable), just need to figure out how to pass it to BrewFather and get it so that you can select your units.

I'll probably work on cleaning the temp comp up over my Christmas break.

quarky42, 2021-01-07:
No big rush, just a very nice to have thing.  Thank you for adding this.   My initial thought would be to have the existing variable output as it is current.   Perhaps a new variable can be output as "gravity-corrected" or "gravity_corrected", something like that.  If it needs to be something short then:  gravitycorrect or gravitycor with or without hyphen or underscore.

MonTage69, 2021-01-12:
Noticed that when cold crashing the gravity goes up. 
20C = 1.011
2.5C = 1.020

Maybe this is purely on the iSpindel side.

Pretty sure my calibration was done correctly as it's always been pretty spot on on my manual readings within 0.001 of a point.

I know the tilt is able to adjust. 

Any pointers?

quarky42, 2021-01-13:
I had something similar to MonTage69, which may warrant posting this as a separate issue, but I respond here because that is where his comment is and would like to know what the devs think about this:  Is this something that might be related to temperature compensation not being implemented (I think this is unlikely.  Temperature should be a -0.002 to 0.002 effect in the range of 20 degrees to 80 degrees F.)  or if what is demonstrated in this graph is a normal kind of behavior to have values take off like that?   

The areas marked with red line / red arrow are where I think there are erroneous values.  One of the red lines is where I took the container out of the temperature controlled bath and put it outside in <25 degree F outdoors to begin cold crashing.  Notice how one of the two iSpindels is reading a big change in specific gravity but it is doing it gradually.  It's not like something got jostled too hard and instantly changed.  As temperature went down from 80 F to 20-something F the specific gravity shot up.  I did notice that the temperature sensor read a low of about 50 degrees. I expected it to read around 30-something degrees.

My actual final gravities as measured by hydrometer were 1.005 and 1.007, but one of the iSpindels gave a much higher result when temperature changed. 

In this graphic, it read accurately for a little bit.  Then they both indicated that the specific gravity shot up (it didn't, but something caused both of them to read that way at first) and then they both acted pretty normal for the batch.  I added 150g more sugar when it got near 1.000, the gravity rose as expected, and then when I cold crashed, one iSpindel was normal and the other one shot up quite a bit again with a big temperature change.

[image]

I am not putting this as a separate issue at this point because I really have no idea what contributed to it and this is just a one-off thing.   When I make another batch, if it happens again, then maybe I can start to say it is repeatable.  At this point I just invite any kind of educated commentary on it so that I may learn more and watch out for more here.

MonTage69, 2021-01-13:
@quarky42 

Yes my temp looks pretty much the same. the gravity is in Red 
[image]

Seems like we have a similar curve. it is much smoother in Brewfather;
[image]

ErikdBr, 2021-01-13:
> At this point I just invite any kind of educated commentary on it so that I may learn more and watch out for more here.

Most substances shrink with a drop in temperature. In other words, the volume will decrease. As a result, the density will increase. So the iSpindel will float less deep in the wort reporting a higher SG if you haven't taken this effect into account in your formula. I think that is what @thegreatgunbantoad wants to build in.

quarky42, 2021-01-13:
> > At this point I just invite any kind of educated commentary on it so that I may learn more and watch out for more here.
> 
> Most substances shrink with a drop in temperature. In other words, the volume will decrease. As a result, the density will increase. So the iSpindel will float less deep in the wort. Thus the iSpindel reports a higher SG if you haven't taken this effect into account in your formula. I think that is what @thegreatgunbantoad wants to build in.

As stated, that effect is -0.002 to +0.002 been 20 and 80 degrees F.  The specific gravity rising almost +0.020 is about 10x the actual amount and if you actually looked at the graphic I posted you'd see that one iSpindel didn't have this effect and the other one did.  What you said doesn't apply.  The increase is too much and only apparent in one of the two graphs.

quarky42, 2021-01-13:
I get that you're trying to help, but I think if you looked a little more carefully at it, you'd see:

My starting specific gravity was nowhere near beer.  It was wine.
They were very close to eachother until the end, so yes they were both grape juice... They were the same, side by side.
One experienced a very high increase of SG / angle change just by getting colder and the other one didn't.  I expected a 0.002 change in my Specific Gravity.  I got 10x that, but only in one sensor.

You're saying things that all have truth to them, but they don't apply to the facts of the situation here that I invested considerable time to try and explain one way or the other.   I'm not upset but I am pointing out the information that I put there already and what it means because there is a lot you didn't see and didn't understand.  Maybe by explaining the nuances here, you'll get something out of this too.  If it seems like I'm upset, I assure I'm not.  I am just explaining the facts as they were presented and some of the more basic conclusions drawn from them.

I had an erroneous spike at the start of the batch too.  It was smaller but still much larger than any kind of Temperature effect, I think.   That time both of them experienced the same spike.  I'm interested in finding out what someone intimately familiar with the iSpindel thinks about this.  It's not a big deal but it is puzzling to me.

thegreatgunbantoad, 2021-01-13:
> As stated, that effect is -0.002 to +0.002 been 20 and 80 degrees F. The specific gravity rising almost +0.020 is about 10x the actual amount and if you actually looked at the graphic I posted you'd [...]

H

thegreatgunbantoad, 2021-01-13:
Ahh poo I didn't mean to close anything

MonTage69, 2021-01-13:
@quarky42 i see you point you are making. 
Wanting to just confirm which firmware you are using, and did you calibrate both at the same time etc.

I didn't think I had an issue with mine as without cold crashing it's perfect. But seems that you are able to have it working correctly with one but not the other. 

I am wondering if I should build another.

quarky42, 2021-01-13:
@ MonTage69 Thank you for taking the time to dig deeper.  That's what I need to try and understand this, I think.

I'm using 6.4.1.   Both iSpindels were calibrated at the same time using the same solutions and I really tried hard to look at the range of the fluttering values.   In other words I didn't just pick any ol' value during calibration but I looked at min, max, and averaged them out over 10 to 15 samples discarding the lowest low and the highest high.   (Sometimes it sucks having a minor in mathematics and a touch of OCD and sometimes it helps.  lol)

I will eventually have 3 units and at some point I will run all 3 units side by side.  It will be interesting seeing if one of them has this odd cold-crash behavior and the other 2 don't.  It will be interesting to see if this is repeatable.   The thing that baffles me the most here is how both of them have that minor "hiccup" early on, but then one of them is fine at the end and the other isn't.  Consistently inconsistent!  :)

quarky42, 2021-01-13:
@thegreatgunbantoad I'm sorry if this ends up hijacking your thread a little bit, but hopefully we find out if this is something worth reporting as its own issue or at the very least, if it is inconclusive, that there is more I can watch out for / try next time knowing that this could happen.  I'm certainly happy to contribute back data / testing of stuff like this in hopes that it can be used to make improvements over time.   If I don't have enough info this time, the result might be that I find out what I need to do better next time.

DesertDog, 2021-01-19:
I have seen the same issue when cold crashing and would appreciate temperature compensation being included.

quarky42, 2021-01-19:
> I have seen the same issue when cold crashing and would appreciate temperature compensation being included.

Temperature itself changes Specific Gravity by -0.001 to 0.002 in the range I ferment and cold crash.  So, there is more going on than just temperature induced specific gravity change.  Maybe the temperature and the pressure inside the iSpindel is throwing off the tilt angle more directly?   If that were the case, coming up with a temperature / pressure bouyancy tilt offset would be hard.  P*V=n*R*T so temperature and pressure are directly related, but how that affects bouyancy and tilt angle I've got no idea how to calculate it.

It is weird that on my side-by-side units one was affected and the other wasn't.

pppedrillo, 2021-01-24:
Here we have a pull request for it #457

thegreatgunbantoad, 2021-01-25:
@pppedrillo ah beat me to formalising my version.  I was holding off pending my other pull request going through.

closed 2021-04-02 by stale[bot]