universam1/iSpindel #438: Very odd stuff going on when trying to add a feature

thegreatgunbantoad, 2020-12-02:
## I'm submitting a ...
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Report

Very odd stuff going on

### Console Logs

This really needs someone actually running the environment

## Steps to Reproduce (for bugs)
Comment out lines 1105 and 1106 in iSpindel.cpp, build and upload.
Then in the AP in Configuration set the Adjustment Calc to 'temp+0ggg' to cause a parsing error.
This causes the device to keep going crashing into deep sleep and it won't serve the AP all.

Do the same with those lines not commented out and it works fine, I'm stumped.

## Context
Trying to add a feature to allow temperature sensor calibration

## Your Environment
VS Code with Platform IO

[iSpindel.txt](https://github.com/universam1/iSpindel/files/5631956/iSpindel.txt)

thegreatgunbantoad, 2020-12-03:
Log File: 
[log.txt](https://github.com/universam1/iSpindel/files/5635714/log.txt)

thegreatgunbantoad, 2020-12-03:
Turns out I was calling this wrong:
    te_expr *expr = te_compile(_poly, vars, 2, &_err);
The 2 should be a 1 since I only have 1 variable.  Doh!

closed 2020-12-03 by thegreatgunbantoad

ErikdBr, 2020-12-03:
Great work!
For what it is worth I noticed a little indentation deficiency
[image]

thegreatgunbantoad, 2020-12-03:
Good spot, wish I'd read that before doing the pull request. 
There is a code prettier in VS code but I cant remember the keypress