universam1/iSpindel #441: cant find ispindle after flashing 6.4.1

streckerja, 2020-12-10:
## I'm submitting a ...
  - [ ] bug report
  - [ ] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [ ] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [ ] Checked the logs for errors in the terminal console ?
- [ ] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

Please replace this line with a brief summary of your issue

### Console Logs

## Steps to Reproduce (for bugs)

1.
2.
3.

## Context

## Your Environment

- Firmware Version used:
- Browser Name and version:
- Service backend you have configured the iSpindel to talk to:

## Features

**Please note by far the quickest way to get a new feature is to file a Pull Request.**

streckerja, 2020-12-10:
i am new arduino and am using the D1 Mini  from the parts list. I have set up
the arduino ide and have successfully run the captive portal example but when i try the ispindle firmware, i don't see an internet connection.  what can i do to get more information on what is happening with the firmware?  I have tried this on multiple D1 boards with the same results.

ErikdBr, 2020-12-10:
Are all the components of the iSpindel attached to the D1 mini? Because before you can run the firmware of the iSpindel you'll have to build the iSpindel completely. You can flash the D1 mini before you build, but you can't run it before you build.

streckerja, 2020-12-10:
Yes, I think.  I’m a little rusty with my soldering skills but I think all is good.

thegreatgunbantoad, 2020-12-10:
Just quick point to make.

Under normal operation you won't see the device active on your network (though you will see evidence it was there in router logs etc.).  Every x seconds (Default 900 = 15 min) it will wake up, dump out data and go to sleep.

First boot up is different, for a fresh board leave it at least 10 seconds to build its filesystem.  It should then start flashing a blue led every second, when that is flashing the access point should become available (can take a minute or so).  Using something like a mobile phone (just works nicely) you'll see a wifi network called iSpindel, connect and when prompted click the sign in thing.

If you have been into the AP and set the wifi up the unit will plough straight into normal operation on repowering.  to get back into to AP double tap the reset on the Wemos.

If by tapping the reset you can't get into a state where the blue LED flashes every second, you probably have a hardware fault.

thegreatgunbantoad, 2020-12-10:
> Are all the components of the iSpindel attached to the D1 mini? Because before you can run the firmware of the iSpindel you'll have to build the iSpindel completely. You can flash the D1 mini before you build, but you can't run it before you build.

Usually the one that causes the problem is the temp sensor, it freaks out if it can't find that.

streckerja, 2020-12-10:
The temp sensor connects to d6 and they seem to be correctly installed. I have 3 completely wired with the same result on all. No flashing blue light.  Is there any diagnostics that can shed light into this problem?

ErikdBr, 2020-12-10:
Did you try the serial monitor in the arduino ide? It is in the top right corner of the ui. If you see nothing there, hit the reset button on the d1 mini.

thegreatgunbantoad, 2020-12-10:
ohh @ErikdBr what SW do you use?  I tend to punt for VS Code.

ErikdBr, 2020-12-10:
> ohh @ErikdBr what SW do you use? I tend to punt for VS Code with Platform IO.

Arduino IDE, you can find it here: https://www.arduino.cc/en/software

streckerja, 2020-12-10:
Yes, but nothing but the garbage line when I reset the unit.

ErikdBr, 2020-12-11:
Then it is clearly a hardware and/or soldering issue.

streckerja, 2020-12-11:
Just went back to the ispindle troubleshooting page and all of my D1 minis have the squared corners. However my D1 minis are just a bit different than those pictured.

ErikdBr, 2020-12-11:
Can you post a picture of the front and back?

streckerja, 2020-12-11:
I forgot to add that it might be a firmware configuration issue and that there are reports that updated firmware may resolve the issue.  I have not located any updates. Anyone know where to find updates?

streckerja, 2020-12-11:
Here are pictures of both front and back.

ErikdBr, 2020-12-11:
> I forgot to add that it might be a firmware configuration issue and that there are reports that updated firmware may resolve the issue. I have not located any updates. Anyone know where to find updates?
> […](#)
> Sent from my iPhone

6.5.0 is the latest version
https://github.com/universam1/iSpindel/releases

ErikdBr, 2020-12-11:
> Here are pictures of both front and back.
> […](#)
> Sent from my iPhone

I don't see the pictures

streckerja, 2020-12-11:
The email I sent had the pictures attached. However, I switched to the BrewFlash loader and found that my D1 requires DOUT.  Flashed with these settings and it came right up. Everything seems to be working so far.

streckerja, 2020-12-12:
Here are the images:
[image]
[image]

I now think I have this issue resolved.  This D1 Mini uses the ESP8285.  I switched FW Loader from the ESP8266loader recommended on the iSpindel page to the BrewFlasher.  The BrewFlasher says to use DOUT for the ESP82285.  Using this, the Fw loaded and started running.  I don't know yet if there will be other issues because of the differences in D1 Minis.

ErikdBr, 2020-12-12:
Yep that is a clone, this is the original:
https://m.nl.aliexpress.com/item/32529101036.html
No stock at the moment.

streckerja, 2020-12-12:
yep....   I also found out that saving settings wont work unless SPI mode is set to DOUT.   Does anyone know how to do this?

thegreatgunbantoad, 2020-12-13:
> yep.... I also found out that saving settings wont work unless SPI mode is set to DOUT. Does anyone know how to do this?

That's odd SPI / I2C is used for peripheral devices.  It's the accelerometer that uses the I2C interface.  Saving settings is completely contained within the ESP8266 board.

Okay after writing that I think you mean the settings in the flashing software, so that's not quite relevant.

If the standard settings when you hit 'Restore default' don't work, you definitely have hardware problems.

I've used clone boards and when they work they generally continue to work.

closed 2020-12-14 by streckerja