universam1/iSpindel #442: Blynk - ISpindel sending rounded specific gravity value

MatthBou, 2020-12-14:
Hello,

## I'm submitting a ...
  - [ ] feature request

## Report
The problem happens when data are sending to Blynk and using the specific gravity.

The specific gravity value is rounded only two digits after the decimal point, which takes away from the precision.
for example :
- specific gravity read by the iSpindel: 1.052
- specific gravity sent to blynk: 1.05

## Solution
Round gravity value 3 digits after decimal point :
sender.add("4", String(Gravity, 3));

Many thanks

nafraga, 2021-01-24:
excuse me , but this string where would be added ? by putty directly in ispindel ? i have the same problem with blynk ,but  i'm not an expert

MatthBou, 2021-01-24:
@nafraga this line must be modified in the iSpindel.cpp file, line 699

nafraga, 2021-01-24:
and where i find it ? if i dowload the firmware and the source code i don't  see .ccp file

nafraga, 2021-01-24:
can change it directly in my ispindel by serial command ? or i have to download the source code ,change the.ccp file and then? thanks

MatthBou, 2021-01-24:
@nafraga no you can't by serial command. Give me your email i send you a modified .bin

nafraga, 2021-01-24:
nobis81.mn@gmail.com         thanks .  which program you use to modify .bin file?

ErikdBr, 2021-01-24:
> @nafraga no you can't by serial command. Give me your email i send you a modified .bin

I suggest you read https://github.com/universam1/iSpindel/blob/master/LICENSE before sending a modified bin

ErikdBr, 2021-01-24:
> [nobis81.mn@gmail.com](mailto:nobis81.mn@gmail.com) thanks . which program you use to modify .bin file?

You have to alter the sourcecode and then compile it into a bin. Don't know which tools they use for iot .cpp.

nafraga, 2021-01-24:
sorry

MatthBou, 2021-01-24:
> 
> 
> > @nafraga no you can't by serial command. Give me your email i send you a modified .bin
> 
> I suggest you read https://github.com/universam1/iSpindel/blob/master/LICENSE before sending a modified bin

oups sorry

nafraga, 2021-01-24:
sorry too

nafraga, 2021-01-24:
but can i say the program to compile the (c++?) file into .bin?

MatthBou, 2021-01-24:
> 
> 
> but can i say the program to compile the (c++?) file into .bin?

Visual Core Studio and platformio

MatthBou, 2021-02-12:
Solved by  #449

closed 2021-02-12 by MatthBou