universam1/iSpindel #443: A recommendation

jsbjx, 2020-12-14:
The 6.4.1 version is very easy to use, so I have a suggestion. Is it possible to add a function. When iSpindel is placed vertically (such as 0-5 degrees), iSpindel does not emit signals and is in a dormant state. This way the device can be turned off without opening the lid. It is more convenient to use.

thegreatgunbantoad, 2020-12-18:
Doing this by software wouldn't work you can't turn it completely off that way, and how on earth would you turn it back on.
It might be more useful to do this by way of a tilt switch in the power line, but this would lead to a reset every time the device was briefly tilted upright.

I did consider using a normally closed reed switch in the power line so you could then place the iSpindel near a magnet to disconnect the power.  You'd also need something like a normally open reed switch in the reset line so you could reset using a magnet.  You'd need to make sure the two read switches were a reasonable distance from each other.

I do have a unit with a reed switch on the reset pins and it works well for that.  no reason it wouldn't work for power too.
I used this type which fit in the tube well.: https://www.ebay.co.uk/itm/DATATOOL-Magnetic-Sealed-Reed-Proximity-Switch-Security-System-2-Pcs/193810870275?hash=item2d20073003:g:0ykAAOSwB~Rf3Lrr
But for power you'd want a normally closed variant.

thegreatgunbantoad, 2020-12-18:
This would work for either dependant on how you wire it, if it will fit: https://uk.rs-online.com/web/p/magnetic-proximity-switches/2897840/

thegreatgunbantoad, 2020-12-18:
Also, using something like this, you can add wireless charging: https://www.ebay.co.uk/itm/QI-Wireless-Charging-Receiver-Charger-Module-For-Universal-Micro-USB-Cell-Phone-/293519854741?var=&hash=item4457258c95
Just make sure you get the right USB type.

ErikdBr, 2020-12-18:
> Also, using something like this, you can add wireless charging: https://www.ebay.co.uk/itm/Qi-Wireless-Charger-Receiver-Wireless-Charging-Adapter-For-Type-C-Android-iPhone/124276236640?hash=item1cef70e160:g:LHsAAOSwbQhfCBhE
> Just make sure you get the right USB type.

I tried that but probably because you have to bend it to get it in a tube it got really hot during charging, the isolation tape I had  on the minus of the battery started to melt.

thegreatgunbantoad, 2020-12-18:
> I tried that but probably because you have to bend it to get it in a tube it got really hot during charging, the isolation tape I had on the minus of the battery started to melt.

I do have one that will wireless charge fine but it takes so long I resort ack to the USB port anyway

jsbjx, 2020-12-19:
I made a board, and the standby power consumption is only 0.07mA, while the standby power consumption composed of three small boards is 3.8mA, which is 54 times worse. So the use time is greatly extended. If judging from the software, when the angle is 0-10 degrees, no signal is transmitted. It should be possible. Hope to adopt my suggestion.
[image]
[image]

thegreatgunbantoad, 2020-12-19:
Ohh where did you get that tube.

Under normal operation the ESP board on the iSpindel is sleer most of the time, it only wakes up every 15 minutes or so.  You could at that point only turn the WiFi on if the Tilt is greater than 10 degrees but the thing generally pretty dormant most of the time.

jsbjx, 2020-12-19:
> 哦，那管从哪儿弄来的。
> 
> 在正常操作下，iSpindel上的ESP板大部分时间处于睡眠状态，仅每15分钟左右唤醒一次。如果倾斜大于10度，那时候您只能打开WiFi，但是通常情况下，大部分时间该设备都处于休眠状态。

I am from China and it is very convenient to process. The shell is processed by the finished tube lathe for testing. The board has been optimized more than 10 times. Finally satisfied.
[image]

universam1, 2020-12-20:
Disagree, putting the Wifi antenna below water level is a perfect faradic cage so not an idea to suggest

jsbjx, 2020-12-20:
> Disagree, putting the Wifi antenna below water level is a perfect faradic cage so not an idea to suggest

The antenna is on top. has a problem?
[image]

universam1, 2020-12-20:
Ok I see everything upside down

closed 2021-02-26 by stale[bot]

ErikdBr, 2021-02-26:
> I made a board, and the standby power consumption is only 0.07mA, while the standby power consumption composed of three small boards is 3.8mA, which is 54 times worse. So the use time is greatly extended. If judging from the software, when the angle is 0-10 degrees, the WIFI signal will not be transmitted. It should be possible. Hope to adopt my suggestion. American Tilt will sleep when placed vertically.
> [image]
> [image]

@jsbjx Where did you get those tubes? Can you provide a link?