universam1/iSpindel #444: AWS IoT support

camsaway, 2020-12-16:
## I'm submitting a ...
  - [ ] bug report
  - [X] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [X] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [ ] Checked the logs for errors in the terminal console ?
- [X] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

Request to add support for sending data to AWS IoT either over MQTT or HTTPS.
https://docs.aws.amazon.com/iot/latest/developerguide/iot-connect-devices.html

### Console Logs

NA

## Steps to Reproduce (for bugs)

## Context
I'm trying to consolidate my brewing automation into AWS IoT.

## Your Environment
NA

## Features
Would like to be able to choose AWS IoT as a destination for sending data. The generic MQTT and HTTP options don't meet the required standards.
https://docs.aws.amazon.com/iot/latest/developerguide/iot-connect-devices.html

I can see there are a couple of previous attempts that have been abandoned/closed/expired. Is there some sort of limitation that prevents this working?

sergioguimaraes, 2020-12-24:
Very good idea!

camsaway, 2021-01-03:
I'm giving this a go. have successfully created a secure MQTT connection to AWS IoT Core with a hard-coded secrets.h file. Now i need to work out how to give users the ability to import certificates through the configuration interface....

if anyone has any advice/suggestions/guidance, that would be greatly appreciated.

camsaway, 2021-01-05:
I've created a pull request for the sender class changes to provide Secure MQTT connections. Once this has been reviewed (and hopefully accepted) I'll look at adding the AWS option but need to know if there are any class changes required first.

closed 2021-03-14 by stale[bot]