universam1/iSpindel #445: Tilt angle inverted

Ulddyret, 2020-12-18:
This is my first post in Github so bear with me if it should have been placed somewhere else

## I'm submitting a ...
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

## Did you follow the general troubleshooting steps first:

- [ ] Follow through the [FAQ](https://github.com/universam1/iSpindel/blob/master/docs/FAQ-en.md) ?
- [x] Checked the logs for errors in the terminal console ?
- [ ] Searched for [similar reports already](https://github.com/universam1/iSpindel/issues) ?

## Report

When using version 6.5.0 the tilt is inverted ie. horizontal is 90 degrees (as normal) but vertical is 180 degrees as opposed to 0 degrees

### Console Logs

No log needed

## Steps to Reproduce (for bugs)

1. Only applicable for https://3d-mechatronics.de builds with reversed gyro. Former issues reported with this hardware was the angle stuck at 60 degrees after calibration

## Context

I have not tried to recreate the polynominal, but I expect it would function still, but would be opposite with increasing tilt resulting in a decreased SG.
It could cause confusion for new users.

## Your Environment

- Firmware Version used: 6.5.0
- Browser Name and version: Chrome and Firefox
- Service backend you have configured the iSpindel to talk to: Ubidots/thingsspeak

I fixed the issue by modifying the tilt calculation in line 859 to:
return 180 - (acos(_az / (sqrt(_ax * _ax + _ay * _ay + _az * _az))) * 180.0 / M_PI);

This was just to let you know about potential upcoming issues with this hardware.

Perhaps an option to "flip" the gyro axis in the software is needed?

thegreatgunbantoad, 2020-12-29:
So long as your tilt still swings 90 deg between horizontal and vertical.  You should be able to account for this in the polynomial anyway.  The polynomial generators probably handle this anyway (This one should be fine with it: https://github.com/universam1/iSpindel/blob/master/docs/ManualPolyGenerator.xlsx).

Ulddyret, 2020-12-29:
Yes, the poly generator will just create an equation based on the "new" readings in the range (180-90) as opposed to the standard (0-90), I just felt like highlighting it in order to help others with the same build.
Obviously it also means that the weights should make it land around 155 degrees in tap water.

closed 2020-12-29 by Ulddyret