universam1/iSpindel #446: Error mounting File System

streckerja, 2020-12-19:
## I'm submitting a support request => It may be a bug or issue.
I am using a D1 mini with an ESP8285 rather than the ESP 8266 the iSpindel instructions are based on.  I was unable to find any of the correct D1 mini boards.  

When i start the iSpindel, I get the following :

16:10:50.839 -> FW 6.3.1
16:10:50.839 -> 2.2.2-dev(38a443e)
16:10:50.839 -> Worker run!
16:10:50.839 -> mounting FS... ERROR: failed to mount FS!
16:10:50.839 -> 
16:10:50.839 -> ERROR config corrupted
16:10:50.839 -> scanning for OW device on pin: 5
16:10:50.839 -> No devices found!
16:10:50.955 -> scanning for OW device on pin: 12
16:10:51.008 -> Found device with ROM = 28 A9 15 D3 39 20 1 3C
16:10:51.008 ->   Chip = DS18B20
16:10:51.909 ->   Data = 1 61 1 4B 46 7F FF C 10 57  CRC=57
16:10:51.909 ->   Temperature = 22.06 Celsius, 
16:10:52.109 -> offsets not available
16:10:52.109 -> Boot-Mode: Deep-Sleep Wake
16:10:52.109 -> 
16:10:52.109 -> Double Reset detected
16:10:52.109 -> ...........
 
note the 4th line where the FS failed to mount.  I don't know if this is a problem with my D1 mini or some other sort of problem.

universam1, 2020-12-20:
Try using the latest firmware version and call factory reset under maintenance menu to format your spiffs. If that doesn’t help your device is bricked

closed 2020-12-20 by universam1

streckerja, 2020-12-20:
I guess that using the HiLetgo WEMOS D1 Mini won’t work with the iSpindel firmware.  Tried both 6.1.3 and 6.1.4.  I get the ‘failed to mount FS’ error message.   Also getting message that one wire device on pin 5 can’t be found. But everything else seems to work.

streckerja, 2020-12-20:
Is understanding correct?  SPIFFS needs a EEPROM attached to the SPI
interface which seems to be lacking on my D1 Mini chip.

Radu-nic, 2020-12-21:
Hi there
I had the same issue using a Wemos ESP-Wroom-02 D1 Mini.  The fix was adding a customized platformIO build option added to the end of platoformio.ini file and build the project using this config.
In my case was like this:

```
[env:esp_wroom_02]
platform = espressif8266
board = esp_wroom_02
; change microcontroller
board_build.mcu = esp8266
; change MCU frequency
board_build.f_cpu = 80000000L
upload_protocol = esptool
framework = arduino
monitor_speed = 115200
upload_speed = 115200
lib_deps =
  ArduinoJson@>5
  ESP Async WebServer
  I2Cdevlib-MPU6050
  OneWire
  DallasTemperature
  RunningMedian
  PubSubClient
  Blynk
  ThingSpeak
```

streckerja, 2020-12-28:
Thank you Radu-nic.  I have updated the config file for my D1 Mini.  However, I am not sure how to re-build the firmware file.

Radu-nic, 2020-12-29:
Hi Streckerja
You should already have installed visual studio code with PlatformIO. If already done go to platformio home page and open this project. You should have a "bee head" icon in the left bar. Click on it. In the upper window (Project Tasks) you should find your newly added build configuration. Select build, upload from their descendants 
Good luck

streckerja, 2020-12-29:
I tried but not well enough,  Haven't worked with this stuff in years!
here is what i ended up with:

|   |   |-- <I2Cdevlib-MPU6050> 0.0.0-alpha+sha.fbde122cc5
|   |   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |   |-- <Wire> 1.0
|   |   |-- <Ticker> 1.0
|   |-- <Hash> 1.0
|   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |-- <Wire> 1.0
|   |-- <I2Cdevlib-MPU6050> 0.0.0-alpha+sha.fbde122cc5
|   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |-- <Wire> 1.0
|-- <Sender>
|   |-- <ArduinoJson> 6.17.2
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <PubSubClient> 2.8.0
|   |-- <Globals>
|   |   |-- <Hash> 1.0
|   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |-- <Wire> 1.0
|   |   |-- <I2Cdevlib-MPU6050> 0.0.0-alpha+sha.fbde122cc5
|   |   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |   |-- <Wire> 1.0
|   |   |-- <Ticker> 1.0
|   |-- <Blynk> 0.6.7
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ThingSpeak> 2.0.0
|-- <tinyexpr>
|-- <WiFiManagerKT>
|   |-- <Globals>
|   |   |-- <Hash> 1.0
|   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |-- <Wire> 1.0
|   |   |-- <I2Cdevlib-MPU6050> 0.0.0-alpha+sha.fbde122cc5
|   |   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |   |-- <Wire> 1.0
|   |   |-- <Ticker> 1.0
|   |-- <MPUOffset>
|   |   |-- <Globals>
|   |   |   |-- <Hash> 1.0
|   |   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |   |-- <Wire> 1.0
|   |   |   |-- <I2Cdevlib-MPU6050> 0.0.0-alpha+sha.fbde122cc5
|   |   |   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |   |   |-- <Wire> 1.0
|   |   |   |-- <Ticker> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |-- <Wire> 1.0
|   |   |-- <I2Cdevlib-MPU6050> 0.0.0-alpha+sha.fbde122cc5
|   |   |   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |   |   |-- <Wire> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <I2Cdevlib-Core> 0.0.0-alpha+sha.460d810002
|   |   |-- <Wire> 1.0
|-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\esp8285\src\iSpindel.cpp.o
[... 107 lines cut ...]
                         ^
pio\src\iSpindel.cpp:790:33: note: in expansion of macro 'OW_PINS'
     my_OWpin = detectTempSensor(OW_PINS);
                                 ^
pio\lib\Globals/Globals.h:71:29: error: 'D6' was not declared in this scope
     (const uint8_t[]) { D1, D6 }
                             ^
pio\src\iSpindel.cpp:790:33: note: in expansion of macro 'OW_PINS'
     my_OWpin = detectTempSensor(OW_PINS);
                                 ^
pio\src\iSpindel.cpp: In function 'void initAccel()':
pio\src\iSpindel.cpp:819:14: error: 'D3' was not declared in this scope
   Wire.begin(D3, D4);
              ^
pio\src\iSpindel.cpp:819:18: error: 'D4' was not declared in this scope
   Wire.begin(D3, D4);
                  ^
*** [.pio\build\esp8285\src\iSpindel.cpp.o] Error 1
o                                                                      o
===================== [FAILED] Took 30.40 seconds =====================

Environment    Status    Duration
-------------  --------  ------------
esp8285        FAILED    00:00:30.400
================ 1 failed, 0 succeeded in 00:00:30.400 ================

streckerja, 2021-01-13:
I finally got the issue solved. 
Issue: HiLetgo D1 mini which uses the ESP8285 chip with 1m built in flash memory. 
I also changed over from SPIFFS to LittleFS.  So I added the following to platformio.ini:
board_build.filesystem = littlefs
board_build.flash_mode = dout
board_build.ldscript = eagle.flash.1m128.ld
board_build.f_cpu = 80000000L

This should also work with SPIFFS by omitting the board_build.filesystem line and not modifying the code to use LittleFS.

Thank you for your help and to Maximilian Gerhardt with the PlatformIO Community for getting this resolved.

Jim Strecker