universam1/iSpindel #450: Local blynk server feature request

JoeThompsonGitHub, 2021-01-01:
## I'm submitting a ...
  - [ ] bug report
  - [x] feature request
  - [ ] support request => Please note that this repositories issues are intended for feature requests and bug reports.

Could there be a way to include local server blynk? I run a local server and dont want to use the blynk cloud to receive my data. Could there be an addition to include a custom server and port please?

JoeThompsonGitHub, 2021-01-02:
I have rectified the lack of feature by editing the blynk libraries which are imported into the .pio folder when the first compiling has happened. Editing \.pio\libdeps\d1_mini\Blynk\src\Blynk\BlynkConfig.h and changing the domain and port to your local server and custom server port. Then editing \pio\lib\Sender\Sender.cpp with a new blynk server IP.
Doing these two things allows for a custom blynk server to be used. Would be good as part of the API dropdown list though to simplify the user experience.

thegreatgunbantoad, 2021-01-03:
I think this was attempted once before: https://github.com/universam1/iSpindel/pull/418

JoeThompsonGitHub, 2021-01-03:
You are right, looks like its been done another way. My solution also works but involves a lot less work,

thegreatgunbantoad, 2021-01-04:
https://github.com/universam1/iSpindel/pull/418 does seem to add the drop down option in.  Never tried it out as I don't use blink.

closed 2021-03-14 by stale[bot]