universam1/iSpindel #454: I2C address mismatch - Acc Test Connection ERROR!

mross286, 2021-01-14:
## I'm submitting a ...
  - [x] bug report

## Report

When a I2C connection test is performed on the gyro board, it lead to an "Acc Test Connection ERROR!" because the wrong address was coded in the library MPU6050.cpp, line 68 "return getDeviceID() == 0x34;".
0x34 equals to 110100, but the right address, as stated in the datasheet from InvenSense (page 15 of the attached pdf), would be 0x68 (1101000).

[MPU-6000-Datasheet1.pdf](https://github.com/universam1/iSpindel/files/5816673/MPU-6000-Datasheet1.pdf)

## Context

Results in 0.00° tilt value permanently shown in the summary screen, everything else works fine.

Replace "return getDeviceID() == 0x34;" with "return getDeviceID() == 0x68;" resolve the issue and the actual tilt is shown in the summary screen.

## Your Environment

- Firmware Version used: All

ErikdBr, 2021-01-14:
Correct me if I am wrong but mpu6050.cpp is part of another project https://github.com/ElectronicCats/mpu6050
iSpindel only uses the library.

mross286, 2021-01-14:
Hi Erik, I think it actually come from https://github.com/jrowberg/i2cdevlib.
Yes, this explain the problem, but I suspect  has nothing to do with "clone" or "fake" MPU chips but a different revision of the same chip, older (before revision3) with 0x34 and newer (from revision 3) with 0x68 (actually 0x68 with AD0 low and 0x69 with AD0 high).

thegreatgunbantoad, 2021-01-15:
I don't think
`return getDeviceID() == 0x34||0x68;`
is quite correct, won't that do a bitwise OR of 0x34 and 0x64 to get 0x74 and then a comparison of that to the returned ID
I think you'd have to use something like:
`return (getDeviceID() == 0x34) || (getDeviceID() ==0x68);`

mross286, 2021-01-15:
Ok, I understand there's something wrong with "who I am" 6bit 117 register on that kind of problematic mpu board.
That 6 bit code should equal 0x34 because bit 0 and 7 are reserved and hardcoded to 0 as stated at page 45 of the attached document.
[MPU-6000-Register-Map1.pdf](https://github.com/universam1/iSpindel/files/5822809/MPU-6000-Register-Map1.pdf)

closed 2021-01-15 by mross286