universam1/iSpindel #455: WEMOS D1 MINI PRO 16M compatibility

michielvtb, 2021-01-16:
Does anyone know if this firmware is compatible with the WEMOS D1 MINI PRO 16M version?
I made 2 Ispindels with this board, but to me it looks like the flashing is unsuccesfull using the NODEMCU flasher.
I have put the settings on 16M, but I am uncertain of the Baudrate, I used 115200. 
after flashing the Ispindel sometimes partly works. 
in Putty I see often the error mounting FS... ERROR: failed to mount FS!
using an older firmware version (5.8.6) makes things better:
One Ispindel goes in to AP mode, but the temperature sensor is not working, and it does only work when the USB is connected, thus not with the battery, although the led flashes when I turn it on. Pushing the reset button a few times does not blink the LED.
The other Ispindel does work on the battery with firmware 5.8.6 but gives temperature -127 degrees and a battery voltage of 0.00Volts.

I have been flashing for several times, with different settings but I am kind a lost now.

Has anyone build succesfully build an Ispindel with the PRO 16M version? And how did you flash it?

kind regards,

Michiel

thegreatgunbantoad, 2021-01-17:
The firmware as is uses SPIFFS for it's filesystem.  But that's sort of deprecated now.  Might be worth looking to see if it supports SPIFFS, the code may need a rewrite to use something like LittleFS.  I'm not sure the board currently used supports LittleFS though.  It is something I'd like to look into though if i ever get some time.

This might help if you up to playing with the code: https://www.mischianti.org/2020/06/22/wemos-d1-mini-esp8266-integrated-littlefs-filesystem-part-5/

michielvtb, 2021-01-17:
Thank you for your answer. In the mean while I also found out that I am the owner of a fake Wemos D1 MINI PRO 16M. This version has a bad flash memory, to save costs I suppose.
I got one of them running with firmware version 5.6.1, but I don't know why, might be luck.
I also ordered an (hopefully) original wemos pro 16M with the Winbond flash memory. I hope this works better. If I can find the time I will try your suggestion on the fake Wemos in the mean time.

ErikdBr, 2021-01-17:
This is the one where the firmware is made for:
https://a.aliexpress.com/_BTGb89
This is your board from the official vendor:
https://a.aliexpress.com/_BPINC5
WEMOS is now Lolin, see also www.wemos.cc

michielvtb, 2021-01-17:
Unfortunately I share the experience with non working boards from other vendors. I wanted to have a Wemos with a ceramic antenna, and the option to fit an external antenna for better gain. So I orderded the PRO 16M version, this one:
[https://nl.aliexpress.com/item/32651747570.html?spm=a2g0s.9042311.0.0.5dcd4c4dULDVfF](https://nl.aliexpress.com/item/32651747570.html?spm=a2g0s.9042311.0.0.5dcd4c4dULDVfF)
But I think it is a fake since there is no RAINSUN stated on the antenna and the flash is not from Winbond, at least this is not stated at the chip. 
I ordered an original Lolin V3 and also a PRO 16M at a local vendor, which should be (more) orignal, at least with the Winbond flash memory.
Hopefully these ones are plug and play, so that I know that my issues are coming from the fake boards.
I will place an update when I have tested the newly ordered boards.

ErikdBr, 2021-01-18:
> Unfortunately I share the experience with non working boards from other vendors. I wanted to have a Wemos with a ceramic antenna, and the option to fit an external antenna for better gain.....

I think you better stick to the lolin v3 and have better gain by installing a wifi extender if at all necessary.

My brews are in a shed in a to a fermentation chamber converted refrigarator, 15 meters from the house plus a couple of concrete walls in between where the router is, So I have a wifi extender in the shed which is much more powerful then the iSpindel. This is what works for me.

michielvtb, 2021-01-18:
> I think you better stick to the lolin v3 and have better gain by installing a wifi extender if at all necessary.
>
> My brews are in a shed in a to a fermentation chamber converted refrigarator, 15 [...]

Yes that sounds like a good plan. I hope to recieve a LOLIN V3 and an orignal PRO 16M this week. Hopefully these will offer a more plug and play solution compared to the hours of unsuccesfull flashing I have had with te fake copies....

michielvtb, 2021-01-22:
I have tried the V3 and PRO16M which I ordered locally and they are plug and play with the latest firmware version! So it is the fake wemos which caused all my problems.
below a picture of the fake/bad one and the good one. I think the flash memory is to blame, the fake one probably has a cheaper one on it which functions badly with the newest firmware. The shop on ali express is now looking if the flashmemory wrong.....
so for anyone who reads this, do not buy the one on the left, buy the one on the right! (with the winbond flash memory)
[image]

michielvtb, 2021-01-23:
It is still not working flawless. I do not get updates in Brewfather, the serial log shows:
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 3.34
*WM: seconds
*WM: Connection result:
*WM: WL_CONNECTED
saving config...
                Failed to mount file system

need to format SPIFFS: 0
Failed to format file system - hardware issues!
Samples:35 min:63.88 max:63.89 time:614
x: 13948 y: 11770 z: -962
Tilt: 63.89
Tacc: 22.57
Volt: 4.87
Temp: 21.00
Gravity: 20.27
IP: 192.168.2.13

calling HTTP
{"name":"iSpindel001","ID":130564,"angle":63.88597,"temperature":21,"temp_units":"C","battery":4.869656,"gravity":20.26525,"interval":900,"RSSI":-66}HTTPAPI: posting
code: 200
{ "result": "success" }

Final-sleep: 900s; RT: 74791
rll▒▒|▒l▒|
▒
l▒
b|▒▒▒▒r▒b▒
b▒▒nn▒lnn▒▒▒
bp▒▒lrlrlp▒n▒
▒
l
▒▒
b
n▒|▒
▒r▒
b▒▒nn▒l▒▒l`▒nn
l`nr▒▒▒n
▒lr▒▒n
▒l▒▒
▒▒▒▒▒l`▒n▒
          FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 EC 24 5 53 20 1 2C
  Chip = DS18B20
  Data = 1 53 1 4B 46 7F FF C 10 2D  CRC=2D
  Temperature = 21.19 Celsius,
offsets not available
Boot-Mode: Deep-Sleep Wake

going to Config Mode
reboot RFCAL
rll▒▒|▒l▒|
▒
l▒
b|▒▒▒▒r▒b▒
b▒▒nn▒lnn▒▒▒
bp▒▒lrlrlp▒n▒
▒
l
▒▒
b
n▒|▒
▒r▒
b▒▒nn▒l▒▒l`▒nn
l`nr▒▒▒n
▒lr▒▒n
▒l▒▒
▒▒▒▒▒l`▒n▒
          FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... ERROR: failed to mount FS!

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 EC 24 5 53 20 1 2C
  Chip = DS18B20
  Data = 1 54 1 4B 46 7F FF C 10 FD  CRC=FD
  Temperature = 21.25 Celsius,
offsets not available
Boot-Mode: Deep-Sleep Wake

Double Reset detected

going to Config Mode
*WM: Scan done

It looks like the:
Failed to mount file system
need to format SPIFFS: 0
Failed to format file system - hardware issues!

might be the problem? is this due to the pro 16M version?

MatthBou, 2021-01-24:
Brewflasher works for me too! Thanks!

michielvtb, 2021-01-30:
I still have an issue with the fake wemos. I found out that it sometimes does not power up from the battery. I need to plug in the USB cable in the Wemos to let it boot, after that it runs fine on the battery. This behavior is not really reproducible. I have 4 ispindels around at the moment and I have been switching them on and off. Sometimes they don't boot, but not all the time. Sometimes removing and inserting the battery when the Ispindel is still on, makes it boot agian without the USB. But most of the times I have to connect the USB for a few seconds.  I have found Github and Reddit posts about an underengineered LDO on the fake wemos, and I think this can be the cause of the problem. I have ordered some LDO's which should be better (500mA max current instead of 150mA), and I will try to replace them. Does anyone else seen this problem?

closed 2021-04-07 by stale[bot]