universam1/iSpindel #456: Can the config portal be accessed once ispindel is connected to wifi?

Garywoo, 2021-01-17:
Once I have entered my home network's SSID and password, the iSpindel goes into operation mode. 
Is it then possible to view the configuration portal once again while my laptop is also connected to the home network, without having to reset the iSpindel and connect to the iSpindel wifi network?

camsaway, 2021-01-17:
I believe that the webserver is only spun up in config mode. Also in normal mode the device only wakes for a very short interval to send data and goes back to sleep to save battery. So I don't think this would be possible.

thegreatgunbantoad, 2021-01-17:
@Garywoo, @camsaway is correct.  The AP only fires up following detection of a double reset.  The reason for this is that serving the AP is a huge battery drain.

It's possible that something could be put into the code to allow you to send a wake up request that would wake it for a brief period when it sends it data, but you'd have to wait up to 15 mins (or whatever it is set to) for a 1 or 2 minute window to grab it and it would be a huge amount of work.  As it is the device only really sends data, I don't think it receives it other than maybe some protocol handshaking.

closed 2021-03-25 by stale[bot]