universam1/iSpindel #458: temperature reading 0,00

guilleiwd, 2021-01-25:
Hi folks, i have some problems with the temperature reading of my ispindel. it was working great but i dont know why it starts to no giving reading o temperature. i have changed the sensor, the diode, an all the resistances. also updated the last version. I have some of them in the jeffrey board and in thepcb 4.0 and i have problems in two of them. do you have any idea whats is happening??
kind regards.

ErikdBr, 2021-01-25:
I saw someone with that problem where the soldering of the temp sensor scratched the isolation of the battery which caused a short circuit of the temp sensor.

guilleiwd, 2021-01-25:
Thanks for your reply. Im afraid that is not the problem in this case, because I have replace the sensor and the soldering was complete flat

ErikdBr, 2021-01-25:
Maybe soldered the sensor in the wrong direction(mirrored)?

guilleiwd, 2021-01-25:
Always I make a double check before use the solder. I don't know it's strange...

ErikdBr, 2021-01-25:
Maybe uou can post a picture? And did you hook it up to the Arduino IDE to see what it logs?

guilleiwd, 2021-01-25:
[image]
[image]
[image]
I'm not very good whit programming I will check how to hook it

ErikdBr, 2021-01-25:
Your D1 mini is V2, I had issues with those, some work some don't. I now only use the official ones from Lolin V3.1: https://nl.aliexpress.com/item/32529101036.html
I also don't like how the 2 resistors are soldered together, it should be one 220K resistor(not 230K)

guilleiwd, 2021-01-25:
Thanks for your tips I'm going to order some. Kind regards mate!!

thegreatgunbantoad, 2021-01-25:
A single 220k resistor is indeed fine, regardless of the fact the PCB silkscreen states 230k.

I tend to agree that your ESP board is probably a duff.  Some of these clone boards are just fine others are iffy.  Were you making in bulk, from a pure cost perspective clones will long term end up slightly cheaper, but a hell of a lot of hassle and wastage barely worth it.  They are also quite troublesome to desolder.

Don't forget to solder the reset contacts in btw, your going to want them.

I like the yellow PCB.

guilleiwd, 2021-01-26:
Thanks for the replay. K have order that WiFi D1. Thanks for the tips guys!!

pppedrillo, 2021-01-26:
> Your D1 mini is V2, I had issues with those, some work some don't.

@ErikdBr That strange. I made vast majority of my iSpindels based on V2 board (exactly as @guilleiwd has) in the past, no issues with any of WemOS V2 boards (about 200 pcs was made) in a course of 2 years. May be my luck tho :D 
MPU6050 modules are another story tho :(

ErikdBr, 2021-01-26:
I saw a D1 mini with squared-off top corners, word goes around that these are the bad ones as stated on this page:
https://www.opensourcedistilling.com/ispindel-troubleshooting/

I recently build one with a D1 mini V2, everything seemed fine in config mode but when in reporting mode it sends a signal to the service, ubidots or brew-spy etc, only once and after that it seems dead.

pppedrillo, 2021-01-26:
> I recently build one with a D1 mini V2, everything seemed fine in config mode but when in reporting mode it sends a signal to the service, ubidots or brew-spy etc, only once and after that it seems dead.

@ErikdBr Its not dead, but rather can not properly wake up from a deep sleep - well-known issue "My ispindel sends data only once". Usually causes either by using a resistor instead of BAT43 diode, or dodge BAT43 diode (yes, where are alot of BAT43 on aliexpress, which parameters are way off the normal BAT43) - partially covered in discussion of issue https://github.com/universam1/iSpindel/issues/59

ErikdBr, 2021-01-26:
I buy my diodes on aliexpress, haven't had an issue with them so far. I will buy some original ones and see if it makes any difference.

pppedrillo, 2021-01-26:
> I buy my diodes on aliexpress, haven't had an issue with them so far. 

@ErikdBr I believe its a lottery like any other shopping on aliexpress. As we already know they are selling sort B- and C- stock of MPU6050 along with good quality. I think its the same with all the rest components and boards. If you lucky enough , you'll get good ones, if not- well.....

ErikdBr, 2021-01-26:
So I allready test the mpu6050's on my Arduino uno as soon as I get a delivery. I will extend this with testing the ds18b20's on the Arduino and a test of the diodes with a multimeter.

ErikdBr, 2021-01-26:
I tested a few bat43 diodes tonight with a multimeter in diode modus. Out of the box they have a voltage drop of 216 and the ones allready soldered have a voltage drop of 230. That is with the plus cord on the anode and the minus cord on the cathode of the diode. The other way around I get a reading of 1. From what I understand that is well within specs. And this was a sexond batch of 50 diodes I bought on aliexpress for € 0.07 a peace including shipping.

closed 2021-04-04 by stale[bot]