universam1/iSpindel #459: Wemos not sending to Ubidots: Acc Test Connection ERROR!

mc1893, 2021-01-25:
Hey!
I set everything up on the Wemos a few weeks ago and it sent the data to Ubidots flawlessly. 
Now I reconnected the Wemos to the laptop a few days ago and unfortunately I no longer receive any data on Ubidots.

So I monitored the output. If I push the reset button, I see the following output:
```
FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
reading config file
parsed config:
{"Name":"iSpindel000","Token":"myToken","Sleep":10,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":200,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"Vodafone-4699","PSK":"xxxxxxxxxxxx","Offset":[0,0,0,0,0,0]}
Acc Test Connection ERROR!
offsets not available
Boot-Mode: External System
power-cycle or reset detected, config mode
going to Config Mode
Acc Test Connection ERROR!
Acc Test Connection ERROR!
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

```

I connected once to the WLan of the Ispindel, where I then got the message that the Wemos could not connect to the WLAN. So I entered the setting again:

[
[image]
](url)

Saving the seetings gives me this output of the wemos:
```
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: iSpindel000
*WM: Parameter
*WM: sleep
*WM: 10
*WM: Parameter
*WM: vfact
*WM: 200.00
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: tempscale
*WM: 0
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 0
*WM: Parameter
*WM: token
*WM: xxxxxxxxxxxxxxxxxxx
*WM: Parameter
*WM: server
*WM: 
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: channel
*WM: 0
*WM: Parameter
*WM: uri
*WM: 
*WM: Parameter
*WM: db
*WM: ispindel
*WM: Parameter
*WM: username
*WM: 
*WM: Parameter
*WM: password
*WM: 
*WM: Parameter
*WM: job
*WM: ispindel
*WM: Parameter
*WM: instance
*WM: 000
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
Acc Test Connection ERROR!
Acc Test Connection ERROR!
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
Acc Test Connection ERROR!
*WM: 3.42
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...
{"Name":"iSpindel000","Token":"xxxxxxxxxxx","Sleep":10,"Server":"","API":0,"Port":80,"Channel":0,"URI":"","DB":"ispindel","Username":"","Password":"","Job":"ispindel","Instance":"000","Vfact":200,"TS":0,"OWpin":12,"POLY":"-0.00031*tilt^2+0.557*tilt-14.054","SSID":"Vodafone-4699","PSK":"xxxxxxxxxxxxxxx","Offset":[0,0,0,0,0,0]}
saved successfully
```

For me it seems like the wemos is now connected again to the WLAN, but I'm still not seeing the Wemos sending any message to Ubidots?
Is the "Acc Test Connection ERROR!" the problem? What is the cause of this error?

Any idea what's going wrong here?

Thanks for helping!

ErikdBr, 2021-01-25:
What does it say on the info page of the configuration, it look like it's not finding the gyro sensor.

mc1893, 2021-01-25:
[image]
This confirms your assumption, doesn't it??

ErikdBr, 2021-01-25:
Yes it does. It is fried or dodgy soldering.

thegreatgunbantoad, 2021-01-25:
@mc1893 I've also had trouble with Ubidots recently.  I have a know perfectly working iSpindel.  Been sending to BF for months.  Decided to attach it to Ubidots for a play and no dice.  I have used Ubidots before, but since it's moved to stem accounts it doesn't seem to want to accept data.

pppedrillo, 2021-01-26:
@thegreatgunbantoad He clearly has tilt readings as zero (see screenshot of Info screen), not a Ubidots problem in any means.

@mc1893 You have dodgy MPU-6050 board, sorry to say that. Just replace it with working one. 
If you see "Acc Test Connection ERROR!" iSpindel will not send any data to anywhere.

daveparrywellmeadow, 2021-01-28:
I've had the same problem. I ordered some spare accelerometer boards (GY-521) and the new ones work perfectly. I am not saying that I haven't caused the problem with my clumsy soldering of the first one - but once they are dead they are dead. I suppose it could have been Dead on Arrival, but who knows now.

The only difference I had was the tilt readout was 'nan' instead of 0.0deg.

ErikdBr, 2021-01-28:
> The only difference I had was the tilt readout was 'nan' instead of 0.0deg.

You will see 'nan' or 0.0 depending on the version of the firmware.

thegreatgunbantoad, 2021-01-29:
@pppedrillo I was referring to this statement "For me it seems like the wemos is now connected again to the WLAN, but I'm still not seeing the Wemos sending any message to Ubidots?"

JoeThompsonGitHub, 2021-02-04:
Has anyone found a site to buy reliable gyros from?

closed 2021-04-14 by stale[bot]