universam1/iSpindel #460: Change WiFi name

kitmir, 2021-01-30:
Hi,
I would like to calibrate multiple iSpindel at the same time.
The problem I face is that all iSpindels have same wifi name and therefore it is not possible to know which one I am connecting.
Only option I see is that I need to open the lid and switch the one I do not use.
Is there anyway I can change the Wifi name from the menu?
Thanks!

ErikdBr, 2021-01-30:
The WIFI name of the iSpindel in configuration mode is allways the same and can't be changed.
So you have to calibrate them one by one.

JochenV1976, 2021-03-10:
Shouldn't the WiFi name be unique for every iSpindel? Somewhere in the doc it's written that the WiFi name should be iSpindel_xxxxxx?

thormj, 2021-04-01:
It is now... we can probably close this for a bit.
I have a patch that puts the "Name" so if it's not been configured it's the node ID, but if it is, it becomes "iSpindel_White" .

closed 2021-06-09 by stale[bot]