universam1/iSpindel #461: Flashing with DOUT instead of DIO

michielvtb, 2021-02-01:
I have a supposed fake Wemos d1 PRO 16M clone which I was not able to flash. The supplier told me to flash it with DOUT instead of DIO. This seems to work, but now I sometimes have startup issues. The board sometimes only boots when the USB is connected and not from the battery. 
Is there any chance that this is because of flashing with DOUT instead of DIO? And if so is it easy/possible to change the firmware?

pppedrillo, 2021-02-03:
What flash chip do you have? I'm guessing it's a chip which only supports DOUT/QOUT in its command set, not QIO/DIO.

michielvtb, 2021-02-03:
It is the GO CHN 25Q128A, the seller in Ali told me it is only usable in DOUT. 
I can program them with DOUT but they sometimes Will not boot from the battery. Thus I was wondering if a DOUT flash memory does not work correctly with the Ispindel firmware.

pppedrillo, 2021-02-03:
If it got flashed successfully, can (at least sometimes) be started normally and after that is working without crashes until poweroff - it is most likely not the flashing issue but rather another yet easter egg in that wemos module.
Did you check your battery btw? Voltage/current it gives to Wemos? I've seen funny TP4056 modules (charger) which cut battery output voltage down to 3.2V to Wemos! :)

michielvtb, 2021-02-04:
Yes the voltage on the Wemos module on the 5V pin is around 4V, depending on the battery voltage offcourse. But when I measure it, it is the same as I measure over the battery itself. 
Another easter egg is very likely, nothing looks genuine on the module :-).
maybe the LDO has not enough output current for startup, I think all the fake ones have 150mA max current which can be critical.

closed 2021-04-14 by stale[bot]