universam1/iSpindel #465: License makes forking not permitted

MarcoCLA, 2021-02-10:
The current license of iSpindel is really restrictive and a bit of an odd duck under licenses. This license is not really "Open Source". That in itself is not bad but it also limits creating derived works.

 >  It's not permitted to distribute the modified project.

So theoretically I am not allowed to fork the code and modify it as it is the same as _distribute a modified project_.

> Modification can be distributed via the official iSpindel release only.

If I do simply not agree with some choices made in the code I am not allowed to change and share them. If @universam1 would for some reason stop accepting pull requests for this software nobody is allowed to make changes to the code and distribute it. This is very bad for a project like this.

I think change to a more permissive license and also an Open Source license is best for long term. If the license is not changed I think there is not long term benefit of adding/fixing for people outside the project as @universam1 has the sole right on distributing these changes. However he does not own the rights to all the code, this is a kinda weird license :yum: .

beikeland, 2021-02-11:
I wouldn't be surprised if at least part of the reasoning for the current terms of the project is to attempt to ensure the Tilt lawyers are kept at bay.

While I understand the concern you raise I wouldn't be too shocked if this thread didn't get a lot of response.

thegreatgunbantoad, 2021-02-15:
@MarcoCLA yeah I was surprised it wasn't just a standard GNU Licence

vytux-com, 2021-02-15:
I'm pretty sure it was against the licensed to sell assembled spindle kits too, but based on comments in this issue tracker quite a few buy parts to make 500 or so units... That's a bloody big homebrew setup if you need 500 spindels to monitor them

gingerneil, 2021-02-24:
> I'm pretty sure it was against the licensed to sell assembled spindle kits too, but based on comments in this issue tracker quite a few buy parts to make 500 or so units... That's a bloody big homebrew setup if you need 500 spindels to monitor them

There is a limit on the license against 'commercial use'. This is hard to define, and doesn't prevent sale of constructed unit. But where is the 'commercial' line drawn? A few units via eBay, a few hundred (thousand?) via an online store?

vytux-com, 2021-02-25:
If you are giving them away sans payment (e.g. free) then there is no commercial sale. If you are getting a payment then it's a sale... It doesn't even have to be real money and can even be at a loss to still qualify as a commercial sale (maybe not a wise one if you are selling for less than the sum of the components but it's still a sale)

closed 2021-05-03 by stale[bot]