universam1/iSpindel #466: Nan message in Tilt and Gravity

RenatoMaia315, 2021-02-12:
I just assembled some of the equipment and in some of them when I access the Info tab, in Tilt and Gravity I get the message "nan". The light comes on and the voltage is ok. Is it a manufacturing defect in the gyroscope module? Any suggestion?

ErikdBr, 2021-02-12:
Nan stands for not a number, the software expects numeric data but gets nothing or alphanumeric data.
Probably a faulty gyro or dodgy soldering. Check your soldering, if that is ok I am afraid the gyro is broke.
There are a lot of clones out there of questionable quality.

RenatoMaia315, 2021-02-13:
Thank you ErikdBr. At first I thought it was some defective solder point, but everything is ok. I rewrote the firmware with a lower speed and finally changed the module, but got the same answer. "nan". 

In some of them, the Gravity field is reset to zero, but Tilt works. I believe it is a lot of defective modules. The first batch (30 pcs) did not show any defect, but in this second batch (50 pcs) so far 10 of them are defective (20%). Both lots were purchased on Aliexpress, but different sellers.

However, the batch of beveled-side Wemos plates, which I imagined to find some defective ones, has so far not been a problem.

jervaters, 2021-02-13:
I have also been dealing with the NaN on the Tilt and Gravity, but funny enough it reports the right numbers to Ubidots. I've had some dodgy gyros on my first build and this is my second now and first succcessful build. I tested the gyro on a breadboard before soldering it to the PCB. 

I'll probably try to build another one now that my soldering skills have improved because I also have an issue where it sometimes works fine, reporting to Ubidots, and other times it just reports the same tilt value over and over again. But a restart once or twice seems to fix it.

JanVorst, 2021-02-16:
I have the same issue. Just build an Ispindel, Checked all soldering. Measured current which is ok. Gives NAN for tilt and gravity in configuration mode, but does give gravity and tilt on ubidots. Does anyone have a solution for this? kinda odd if you ask me.

RenatoMaia315, 2021-02-17:
JanVorst,

If you don't have the equipment to change the MPU6050 chip, the only solution is to change the module.

The TDK-making company has discontinued the MPU6050 chip and discourages new developers from continuing to use it. The newest version is from the ICM family.

I ended up assembling a device just to test the modules that worked and the total of 50 pcs purchased on Aliexpress from the seller Win win 22 went to waste.

Commercially, the seller acted correctly. The question remains, does he know that he is selling defective products?

JanVorst, 2021-02-17:
RenatoMaia315 thanks for your reply. 

Well... that kinda sucks for me. I am not capable to change out the chip, and apparently i cannot get my hands on an new module cheap. MPU6050 are everywhere but the ICM-20689 (as i understand) is only available with large shipping. 

can i calibrate the Ispindel with unidots? It gives values there.

ErikdBr, 2021-02-17:
And is that even going to work as ICM-20689 is a 6-axis gyro opposed to the MPU6050 which is a 3-axis?

JanVorst, 2021-02-17:
They say you can replace it directly:
https://blog.cdiweb.com/2017/07/21/tdk-invensense-nrnd-replacement-parts/

But that is not a solution for me. What I find curious, is that the tilt works fine for unidots. Which still makes me think it is not an hardware issue. But I am not an expert on this matter. MPU6050 has been used alot over the years right? Has this problem ever been an issue for the iSpindel?

I still have a question to work around this:
can i calibrate the Ispindel with unidots? It gives values there.

JanVorst, 2021-02-20:
i did a test on the module on 5v with my arduino, seems to giva an address. So i don't think the hardware failed.

[image]

pppedrillo, 2021-02-21:
@ErikdBr It can also display NaN if there was a division by zero in calculation which in turn were converted to string.

ErikdBr, 2021-02-21:
> @ErikdBr It can also display NaN if there was a division by zero in calculation which in turn were converted to string.
> **agreed**
> And looks like NaN appears only with using new "improved" formula for tilt calculation. **I build a couple with the new tilt calculation version 6.5.1 and experienced no issues so far**

pppedrillo, 2021-02-21:
@ErikdBr It is for dodgy gyro only. Before "improvement" UI was showing "0". Now it shows "NaN". 
@thegreatgunbantoad  may be you can add another yet "if-else" there into your "new" calculations?

JanVorst, 2021-02-21:
I did some more experiments with arduino. if i use the arduino my voltage for the module is around 3.20 to 3.30. If i look at the voltage on my board it is 3.6. I use a 220 ohm %5 resistor which could potentialy carry 230. But my hunch now is that you really need a 230ohm resistor, or modify a 220 with a 10 ohm resistor. Although i hear alot of user not having an issue with 220ohm only.

I tried powering the module with the ispindel and read with the arduino. It gives nothing when i power it with the ispindel. My technical knowledge isnt that good, so i dont know if this could be the issue and if my setup is corect. But no reading the i2c or gyro when powering the module with the ispindel. Did i found the issue? My test setup is shown below:

[image]

ErikdBr, 2021-02-21:
I am noy exactly sure where you are using that resistor but I use a BAT43 schottkey diode between RST and D0, and a 220K and a 4.7K resistor, both kilo-ohms!

JanVorst, 2021-02-21:
Yea i use kilo-ohms too my bad. i use a 4.0 pcb board, i bought a kit. There was only a 220kilo-ohm resistor with it not a 230 kilo-ohms

https://www.pcbway.com/project/shareproject/iSpndel_4_0.html

JanVorst, 2021-02-21:
@ErikdBr, I feel like a doofus. I checked it again... its a 220 ohm resistor, not a 220K ohm resistor. Dear lord, that is a problem for sure. I will try changing out the resistor. It is a wonder the module still works on android. i will keep you updated.

ErikdBr, 2021-02-21:
Lol

pietro-22, 2021-02-22:
I’ve been having this issue as well. I believe my resistors are all within check, but while running putty, both of my ISpindels are returning “Accel Test Error” repeatedly. I’ve gone through reading a couple of the other gyro issues posted within here and it appeared that one user altered the code by commenting out part of the void flash calibration code. I am a novice in this area and have not been able to recompile the code to see if this is what’s causing my error. For reference, I bought my GY-521s from the hiletgo store on Amazon. Anyone have any idea how I can get around this issue before replacing the gyros? Thanks in advance!

ErikdBr, 2021-02-22:
I suggest you test you're gyros before soldering as I described here: https://www.homebrewtalk.com/threads/ispindle-diy-electronic-hydrometer.598187/post-9054316

Further more the clones of the Wemos/Lolin boards are giving lots of problems and might or might not work, usually they don't it is a waste of money!!!!!!
Don't use these clones buy the original ones, see https://www.wemos.cc/en/latest/d1/d1_mini.html

pietro-22, 2021-02-22:
@ErikdBr I did see your post on homebrewtalk, but wish I thought of that before soldering the gyro. Also, I am assuming that my Wemos board is a clone and that could also be the issue. Unfortunately both are already soldered. Any idea of what else I could do? Is it more likely that the boards are bad or something I could fix by changing a line of the code?

JanVorst, 2021-02-22:
I tried commenting the flash line out. It worked for the part that i could see the tilt. Even with the dingy resistor. But calibration did not work, it stalled. But hey, I used the wrong resistor.  

You could try it too, i made a build. It is in the zip. You can just try it out without all the firmware build hassle. 

[firmware6.5.1 JVV.zip](https://github.com/universam1/iSpindel/files/6024686/firmware6.5.1.JVV.zip)

RenatoMaia315, 2021-02-22:
> I suggest you test you're gyros before soldering as I described here: https://www.homebrewtalk.com/threads/ispindle-diy-electronic-hydrometer.598187/post-9054316
>
> Further more the clones of the [...]

My report. All parts purchased from Aliexpress stores. 50 pieces of each.

Wemos (clone) - 1 defective

TP-4056 - 2 defective

GY-521 - 22 defective - NAN

GY-521 - 2 defective, incorrect angle measurement. Should measure 90 degrees horizontally, but one measures 76 degrees and the other 83 degrees.

BAT43, DS18B20, Resistors, Board Joey Joe Joe Jr V2.1 (PCBWAY) - 0 defective

That´s all folks

ErikdBr, 2021-02-22:
> Wemos (clone) - 1 defective
Could you provide us with a link to these, cause all of the clones I bought didn't work ?
(Btw if you follow the buy it link from the wemos.cc link it leeds to the Lolin store on aliexpress)

RenatoMaia315, 2021-02-23:
Yes guy...

https://www.aliexpress.com/item/32651747570.html?spm=2114.13010708.0.0.7cc6b90a6xe4Gt

Ok?

pppedrillo, 2021-02-23:
@ErikdBr @RenatoMaia315 
It is useless to keep a registry of chinese online shops.
Today they sell great ones, tomorrow dodgy C-stock, next week - again a good ones.... etc.

RenatoMaia315, 2021-02-23:
It seems to me that there is no quality control. It's a lottery ... In another purchase (30 pcs - red led), no gy-521 was defective. In this shipment (green led - 50 pcs) 45% came defective.

ErikdBr, 2021-02-23:
> @ErikdBr @RenatoMaia315
> It is useless to keep a registry of chinese online shops.
> Today they sell great ones, tomorrow dodgy C-stock, next week - again a good ones.... etc.
> 
> Buying there you always pushing the luck unless you have an option for return or replacement for free of charge (but even then in case of aliexpress it will take enormous amount of time).

> It seems to me that there is no quality control. It's a lottery ... In another purchase (30 pcs - red led), no gy-521 was defective. In this shipment (green led - 50 pcs) 45% came defective.

Agreed, it's a lottery except for the original Lolin D1 mini's they allways work, I have had no failures up until now of the 32 I bought. Yes they cost one euro more but I don't think that's a problem.

I too had GY-521 with red led(all good) and green led(lot's off failures, also lot's are good)

pietro-22, 2021-02-24:
So, if altering the source code doesn’t work, I suppose my only other option is to buy new GY-521s. Where would you say is the most reliable source to purchase?

ErikdBr, 2021-02-24:
> So, if altering the source code doesn’t work, I suppose my only other option is to buy new GY-521s. Where would you say is the most reliable source to purchase?

The last ones I ordered that worked are:

https://www.banggood.com/5Pcs-6DOF-MPU-6050-3-Axis-Gyro-Accelerometer-Sensor-Module-p-959259.html?cur_warehouse=CN - red led

https://nl.aliexpress.com/item/4001203301527.html - green led

JanVorst, 2021-02-24:
I ordered some extra gyros because even with the right resistor, i get the same message that the connection failed. I even walked through the code, but at the moment there is a registry assignment the gyro stalls. So no calibration possibilities. That is kinda silly, because the gyro it self and the i2c works. funny that the resistor didn't change much. Hope it didn't break anything. Anyway i have two things that I want to ask you guys:

1. It kinda sucks that the mpu6050 registry is so untrustworthy. I wonder if there an alternative gyro which could be used instead? One that is more reliable? 

2. Aside from that, i think that my gyro would have been useful, even with the registry fails. If I read this post below, someone is having the same issue, but fixed it by converting the raw data and not the offset which you need the registry for. I am not capable enough to do this in the firmware, although i think this could be a practical, for more... cheaper solutions. Do I have a point? And if not, why not?

https://forum.arduino.cc/index.php?topic=680775.0

ErikdBr, 2021-02-24:
Maybe this shines a light: issue #454

JanVorst, 2021-02-25:
I ordered some from Conrad and ali express. There not that expensive anyway.

ErikdBr, 2021-02-25:
> I ordered some from Conrad and ali express. .....
Let us know how that works out!

pietro-22, 2021-02-26:
@ErikdBr thank you! However, before I purchase anything new. I am having trouble recompiling the source code. I am able to open the zip file and edit it in the platformio IDE, but I cannot understand how to compile the code to then upload the .bin to each of my ISpindels. Sorry, I am a very novice coder...Thank you!

JanVorst, 2021-02-27:
@pietro-22 i tried all that and it still did not work. Every registry call turns into a fail. If you want to try that is suggest to search how. one tip i can give you, if you build just search (ctrl+p) for the firmware.bin in visual studio code. The firmware will be available in the d1 mini folder. The rest you can find in the instructions. No point in repeating all that stuff.

JanVorst, 2021-03-03:
Alright, got new gyro's and... they work! bought them from the Dutch Conrad site. All three gyros work. Now to desolder the components and solder the right ones in. Wish I tested all of them before i soldered them to the board. Now i have to buy a desolder device, my desolder device was broke. Man, Still shit out of luck. But atleast I have a solution in sight.

JanVorst, 2021-03-08:
Well, buying another gyro worked. I soldered it on the board and it works fine now. If you guys like i can make a photo of the gyro that worked and which did not. One tip, buy from a trustworthy dealer of electronics. 

It's a bit offtopic, but 20 degrees tilt is fine for plain water? I read that it should be 25, but i cant get the board higher in the container. If this is ok, the ispindel is gonna work for me.

pppedrillo, 2021-03-08:
> It's a bit offtopic, but 20 degrees tilt is fine for plain water?

@JanVorst You'll be fine.

ErikdBr, 2021-03-08:
> I read that it should be 25

It should be 25 plus or minus 5. So anything between 20 and 30 is fine, the sweet spot is 25. I have some less then 20, like 18 or so and they work fine too. I even have one with 9 in plain water, I use this as an advanced thermometer 🥇

pietro-22, 2021-03-11:
@JanVorst i found another supplier on Amazon: 
https://www.amazon.com/gp/aw/d/B07MMZ37PT/ref=ox_sc_act_image_1?smid=AYZI0TO4JGBW9&psc=1

Do you think it is a good move to buy from this source?

JanVorst, 2021-03-11:
@pietro-22 The good news is that these look like the ones i have that worked. It got a yellow resistor on it ad the side and it reads GY-521. 

The bad ones are black with ITG/MPU printed next to them. green led.
like these: 
https://www.amazon.com/HiLetgo-MPU-6050-Accelerometer-Gyroscope-Converter/dp/B01DK83ZYQ/ref=sr_1_1_sspa?dchild=1&keywords=GY-521&qid=1615493460&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzSzkwVTlBNTdMWUc4JmVuY3J5cHRlZElkPUEwNzMxODM5MThRNzg5SlQ0OU5FWSZlbmNyeXB0ZWRBZElkPUEwNzgzNzAyMjFCSUM1NVoxSTUyWSZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

But hey, I cannot guarantee that these yellow GY-521 will work.

pietro-22, 2021-03-11:
@JanVorst the current ones that I have soldered have yellow resistors, but they do not have GY-521 printed. They have ITG/MPU. I bought these non working ones from the Hiletgo Amazon store. I think I will purchase the set I quoted above and test them by soldering before I scratch off the led.

holzachr, 2021-05-13:
See my following enhancement to the MPU6050 library:

https://github.com/ElectronicCats/mpu6050/issues/21

Once the iSpindel is recompiled to use the lates version of the MPU 6050 library, it "could" go well for a lot of devices.

wilbrod, 2021-05-27:
> See my following enhancement to the MPU6050 library:
> 
> [ElectronicCats/mpu6050#21](https://github.com/ElectronicCats/mpu6050/issues/21)
> 
> Once the iSpindel is recompiled to use the lates version of the MPU 6050 library, it "could" go well for a lot of devices.

How do we go about getting the iSpindel project to include your fix in the official firmware?

holzachr, 2021-05-27:
At best, @universam1 updates his libraries and recompiles+releases for everyone here.

peterschrott, 2021-06-04:
I also get the NAN for the gyro in the WebUi. On the serial monitor I see: `Acc Test Connection ERROR!`

Nevertheless when I press the rest on the D1 mini there is an gyro value printed:

```
woken from deepsleep, normal mode
Samples:42 min:73.07 max:122.78 time:752
x: 14888 y: 7932 z: 128
Tilt: 89.66
Tacc: 41.44
Volt: 4.30
Temp: 25.63
Gravity: 33.39
IP: 192.168.0.181
````

So, not quite sure if it's broken hw.

closed 2021-08-14 by stale[bot]